ArduPilot/ardupilot #12694: Arduplane 4.0 won't connect to external GPS Mateksys F405-Wing

dposea, 2019-10-29:
I've been trying to get a GPS unit to work for several days. Finally it's working but I had to disable GPS on Serial 3. With Serial 3 set to GPS protocol (5) the GPS connected to Serial 4 was not recognized. I also tried connecting to Serial 3 but that did not work either, and I don't know why. I believe I also changed GPS_TYPE2 from 0 to 2 prior to changing Serial 3 to -1. I think with Serial 3 protocol set to GPS, as well as Serial 4, the only GPS connected is recognized as GPS 2, but I'm speculating.

Version was Plane 4.0, also 4.10 dev

rmackay9, 2019-10-29:
@dposea, In general these reports should go into the [Plane support area first](https://discuss.ardupilot.org/c/arduplane/plane-4-0) and then once it's confirmed to be a bug it can be raised as an issue.  Sorry to be strict but if we're not then our to-do list turns into a support forum and we lose track of what we should be doing.

I'm going to close this if that's OK and then it can be re-opened later if it's confirmed to be a software issue.

closed 2019-10-29 by rmackay9

dposea, 2019-11-08:
Ok, I didnt want it to be missed because it was very frustrating. I wont open an issue unless I can recreate it consistently.