ArduPilot/ardupilot #12704: Preflight_Calibration as RCx_option

EERomeo, 2019-10-29:
### Feature request

As it is advised, it is best practice to do a Preflight_Calibration after a few minutes from power up, to allow the sensors to warm up. As it is now, there is no way to do this procedure if you are not connected by telemetry to a GCS.

I was thinking that we could have this function as an RCx_option so we could assign a spare channel on the remote for when you don't have telemetry for whatever reason.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2019-10-29:
On Tue, 29 Oct 2019, EERomeo wrote:

> As it is advised, it is best practice to do a Preflight_Calibration after a few minutes from power up, to allow the sensors to warm up. As it is now,
> there is no way to do this procedure if you are [...]

Calibrating what?

Gyros would be the obvious one, but there's a few other things you can 
preflight-calibrate - airspeed sensors, ahrs trim and baro spring to mind.

rmackay9, 2019-10-30:
As a side note, is this advice to re-do the calibration on the wiki somewhere?  I personally never do a re-calibration a few minutes after startup.  In very cold environments I can see that this might be necessary... just wondering where this advice is coming from.

proficnc, 2019-10-30:
We give this advice to users. The Gyro calibration when cold can be quite a way off, so we recommend warming the cube up and resetting before flight.

An automated method would be awesome

EERomeo, 2019-10-30:
I don't recall if I've seen this advise on the wiki, but it appears on the community forms.
Now, personally I have noticed a difference for the airspeed sensor.

rmackay9, 2019-10-30:
It definitely sounds useful to me for the airspeed sensor.  .. and I'd say for the gyro as well for dealing with cold temperatures or for when taking off from boats.  I'd guess we can skip the baro calibration..

SergeyBokhantsev, 2019-10-30:
What if to have an option that will periodically run gyros re-calibration (say every 30 seconds) if vehicle is disarmed and is steady? That way we don't need nor for a telemetry connection nor dedicated RC switch. Does it make sense?

EERomeo, 2019-10-30:
I don't think this would work, the vehicle might be moved by the wind. Also, you have to cover the airspeed for the calibration, what if you take the cover off and than it calibrates by itself?
This needs to be precisely controlled and only done when you know some conditions are met.

rmackay9, 2019-10-31:
@EERomeo, right, so the wiki was referring specifically to the airspeed sensor.  That makes sense and it seems like maybe there's a consensus that that would certainly be a good RC Option to have added.

I'm unsure if automatically running the calibrations periodically before takeoff is a good idea or whether it will cause as many problems as it fixes.  The user could pick-up the vehicle for example.  In the case of the airspeed sensor the user needs to partially cover the end of the sensor it seems.

Pedals2Paddles, 2019-10-31:
I would actually suggest a stick gesture rather than an aux channel to initiate the gyro calibration. That would let this work for anyone, even those who don't have a myriad of extra channels.   We already use a stick gesture for compass calibration too, so there's a precedent for it.  Perhaps a single stick gesture could kick off the gyro, airspeed, and baro calibration as one single _pre-flight sensor calibration_.?  The compass cal is both sticks down and in.  So maybe both sticks up and in for pre-flight calibration?

Personally, I keep INS_GYR_CAL disabled, so it doesn't try to calibrate on boot at all.  The only time it does a calibration is during the 3D accel cal process, or if I specifically request it from the GCS.  IMO, this locks it into a known, good, warmed up calibration and eliminates any non-sense it gets when booting up cold.  but a gesture to redo it before flight would still be ideal.

rmackay9, 2019-11-01:
I'm not personally a big fan of stick gestures because some users will accidentally trigger them.  I also find them hard to remember..

VDLJu, 2019-12-12:
> What if to have an option that will periodically run gyros re-calibration (say every 30 seconds) if vehicle is disarmed and is steady? That way we don't need nor for a telemetry connection nor dedicated RC switch. Does it make sense?

I think this would be a good enough solution on multicopters and it could be default off, but enabled by parameters. For example it could check that there is no acceleration, drone is relatively level and low rate acceleration on gyros while calibrating the gyros.

We have seen that some gyros go bad on >15C temperature change, which is quite normal on low ambient (-5C to -25C) temperatures, while there is +4m/s wind and non fully sealed enclosure.

Pedals2Paddles, 2019-12-12:
But if the whole point of the calibration is to learn what perfectly still is, how can we use the gyros to determine it is perfectly still in order to automatically calibrate at interval? That's kind of putting the cart before the horse.  It is manually initiated now because the user has to first ensure the vehicle **is** perfectly still regardless of what the gyros actually think.

rmackay9, 2019-12-13:
It might be possible to ask the EKF how close it is to hitting the gyro bias limit and then check the accels to see that they're not changing and then trigger a re-calibration... I wouldn't be in favour of an auto recalibration for everyone because most situations don't need it.. but if we can detect the special conditions where it would help it might work out.

VDLJu, 2019-12-13:
I have been hitting "Gyros inconsistent" warning quite often on cold weather while waiting the IMU to warm up. It would be great to have re-calibration triggered a bit before getting that far. IMO re-calibration in that case causes less harm than it fixes.

Also if the IMU is warmed up and then rebooted, the "gyros inconsistent" error happens occasionally on windy cold weather where the wind cools down the IMU.

EERomeo, 2019-12-13:
I do not think an automated recalibration would be viable due to the myriad of scenarios when this would end up doing more harm than good. I think only you as the user/pilot know when the conditions are ideal for performing this.
An rcoption or stick gesture is the better approach. I know I need it because my airspeed is reading constantly 4m/s higher after warm-up which is more than I'm comfortable with.

rmackay9, 2020-03-09:
This request appeared on the forums again here: https://discuss.ardupilot.org/t/preflight-calibration-as-rc-option/53327

peterbarker, 2020-03-09:
Perhaps this might be done better as telemetry scripts?

rmackay9, 2020-03-09:
@peterbarker,.. by telemetry script I guess you mean Lua script?

peterbarker, 2020-03-09:
> @peterbarker,.. by telemetry script I guess you mean Lua script?

Yep.  @yaapu

auturgy, 2020-03-09:
Onboard lua perhaps, but I don’t think that implementing this feature via frsky telemetry is the best way forward: it really should target the general use case, not just those with a particular type of RC.

Regards,

James

andyp1per, 2020-03-09:
To be clear - I believe this is asking for pre-flight accel calibration because we see multiple instances of those acquiring large offsets some time after the FC being powered. It's particularly bad on small 1Mb boards with ICM 20689 (so no scripting). My understanding is that gyros are already calibrated at arming - it's only the accels that are not. Obviously the accel calibration function currently involves putting a copter on all 6 axes, so not very convenient for an RCX option. My thinking is that that we would have to do something like assume the copter is largely calibrated and largely level when the option was selected and then make the accel offsets consistent to some frame of reference - not ideal but better than the status quo.
The copters with these problems have extreme cases of the leans and my suspicion is that it is also related to low board voltage.
@1wick is the expert on this kind of the leans!

EERomeo, 2020-03-09:
Actually this is requested for Airspeed sensor mainly, maybe gyros as well. You should not have to do accel calibration on the field, and I don't think the Preflight_Calibration button from Mission Planner has anything to do with accelerometers.
The main concern for me at least is the Airspeed sensor, I get about 4-5 m/s reading by the time it warms up a little and that is a little much and I have to do the Preflight_Calibration to eliminate this offset.

peterbarker, 2020-03-09:
OK, say we narrowed the scope of this to just airspeed calibration.  I could see that - it's a relatively simple option which we fire off from mavproxy with a simple command with no arguments, so the most common use case would seem to fit on a switch.

You now have another "kill vehicle" option on a switch, you do realise that, right?

rmackay9, 2020-03-09:
@peterbarker, re "kill the vehicle", I guess it would get a safety check to ensure it's not flying at the time the calibration is attempted.

1wick, 2020-03-09:
Hi, 
In this case, (responding to Andy's post above), it is the Gyro's I'm interested in.  
I don't know the cause yet..  I can let the copter warm up, then reboot to get a fresh gyro calibration.  But still, as soon as I apply power to the motors, the center line for the gyro (seen in the graphs under IMU-GX) goes above zero and stays there even after I land. (Then slowly goes back to zero after some minutes).  It's almost always at .04.  The copter leans.  It is instant..  I spin up the motors, the gyro gets that offset.  
So, this weekend, I discovered I can take off and hover for literally just a second.  Then land and immediately do a preflight calibration from MP, and then the copter is fine..  .04 becomes zero, and the copter performs perfectly.   
I know this is not something others are experiencing, so don't expect anyone to make this change for me.. I was just curious if others had the idea of making preflight cal an RC option, and if it would be a lot of work to do it..  I'm glad to see there is a bit of discussion about it. 

Off topic, but I'll mention,  I think Andy is correct that it's something to do w/ power.   I have four copters that are all doing this.  different FC's, different frames, different ESC's.  But the one thing that is the same is they're all 2s. All the same (or similar) motors and 3" props.    I've got bigger copters, 5"-16" props, 3s-6s batteries, and they are all fine..

andyp1per, 2020-03-09:
@1wick so what about if we did a post-arm calibration in the same way that we do a pre-arm calibration? That might solve your problem.

1wick, 2020-03-09:
Yes, that might solve my problem.  But probably isn't useful to anyone else.  At least a few others are interested in being able to trigger the preflight_cal w/out having a computer in the field.  

I am curious though, if the offset happens when motors are slowly spinning at their idle speed or if I need to give it more power and take off..  I'll do some testing later.  If I find anything interesting, I'll post over on the main forum.

Andrey-Svetlichny, 2020-10-15:
I use a Matek analog airspeed sensor and usually don't have a computer in the field. After warming up for 5 minutes and running Preflight_Calibration I have average airspeed in the still air less then 2.5-3km/h. If I just disconnect battery and connect it back to run Preflight_Calibration I have average airspeed 12-15km/h. So it will be really useful to be able to trigger Preflight_Calibration from the radio (or just calibrate the airspeed sensor).

stevegut78, 2022-11-14:
Any traction on this? It would be nice to have preflight or (at a minimum) airspeed calibration. I've been powering on and letting warm up then pulling the battery and rebooting to get a good airspeed calibration without a GCS.

Enggmaug, 2023-03-19:
I also am interrested. Wanted to open a new thread when I found this one.
Interrested in the Airspeed mainly.