ArduPilot/ardupilot #12706: Please add dd/mm/yyyy to the OSD panel

ninja-zx11, 2019-10-29:
### Feature request

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[ * ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
Please add dd/mm/yyyy to the OSD.It will be useful while watching achieved FPV recordings.

peterbarker, 2019-10-29:
Support for this was added with https://github.com/ardupilot/ardupilot/commit/1f9b2f712ee7698914d8ed937eb2e7c432c4d587

Should be out in the next version of Copter.  This is not in the current stable release of Plane, but perhaps could be backported.

vierfuffzig, 2019-10-29:
the respective commit https://github.com/ArduPilot/ardupilot/commit/1f9b2f712ee7698914d8ed937eb2e7c432c4d587 actually adds time of day (hrs:mins) only, as i considered that useful and relevant if you‘re under the goggles, in addition to the existing FLTIME panel. 
honestly, i personally am not a fan of adding an 8-digit panel that contains static and redundant information without any primary benefit for inflight use, but i can well propose a respective PR and let the reviewers decide.

peterbarker, 2019-10-29:
On Tue, 29 Oct 2019, vierfuffzig wrote:

> use, but i can well propose a respective PR and let the reviewers decide.

Perhaps flashing the date up on arm and disarm?

ninja-zx11, 2019-10-30:
@vierfuffzig  Thanks basti for your work.Having time on OSD is quite useful!!

@peterbarker Yes that would be great.Even flashing date once while arming will be useful while watching FPV recordings later.

atshort, 2019-11-08:
> On Tue, 29 Oct 2019, vierfuffzig wrote: use, but i can well propose a respective PR and let the reviewers decide.
> Perhaps flashing the date up on arm and disarm?

Like the stats screen when Betaflight disarms. All the things that are really useful to have preserved in the video, but not cluttering the screen during flight.