ArduPilot/ardupilot #12712: Setting EK2_IMU_MASK = 0 cannot be undone

rmackay9, 2019-10-30:
As part of testing this PR:https://github.com/ArduPilot/ardupilot/pull/12710 I found that once EK2_IMU_MASK is set to zero it is seemingly impossible for the user to restore the parameter because EK2_ENABLE is automatically forced to zero meaning the EK2_IMU_MASK parameter will no longer be displayed in the GCS.  A user's attempt to reset EK2_ENABLE = 1 is persistently overwritten with 0.

I could not found a work-around meaning the user might be forced to backup their params, reset to factory defaults, then apply some of the parameter settings again.

kd0aij, 2019-10-31:
I can't reproduce this in SITL:

STABILIZE> param show EK2_ENABLE
STABILIZE> EK2_ENABLE       1.000000
param show EK2_IMU_MASK
STABILIZE> EK2_IMU_MASK     1.000000
arm throttle
STABILIZE> APM: Arming motors
Got MAVLink msg: COMMAND_ACK {command : 400, result : 0}
ARMED
disarm
STABILIZE> APM: Disarming motors
Got MAVLink msg: COMMAND_ACK {command : 400, result : 0}
DISARMED
**param set EK2_IMU_MASK 0**
STABILIZE> reboot

APM: Initialising APM
Init ArduCopter V4.0.0-dev (a19dd63d)
Free RAM: 131072
FW Ver: 120
load_all took 11661us
0 0 0 AP_Logger_File: buffer size=51200
APM: Calibrating barometer
APM: Barometer 1 calibration complete
Init Gyro***
Ready to FLY APM: GPS 1: detected as u-blox at 115200 baud
Time has wrapped
('Time has wrapped', 6068, 216156)
APM: PreArm: EKF attitude is bad
APM: PreArm: EKF2: EK2_IMU_MASK is zero
**param set EK2_IMU_MASK 1**
STABILIZE> APM: EKF2 IMU0 initial yaw alignment complete
APM: EKF2 IMU0 tilt alignment complete
arm throttle
STABILIZE> APM: Arming motors
Got MAVLink msg: COMMAND_ACK {command : 400, result : 0}
ARMED
APM: EKF2 IMU0 origin set
APM: EKF variance

rmackay9, 2019-10-31:
@kd0aij,

When I follow the steps you've outlined I actually get a floating point exception.

The difficulty may also come from the GCS being used.  Maybe try it using MP's built-in simulator?  I think you'll find you can't update the EK2_ parameters because they're not visible because MP can't get them from AP.

kd0aij, 2019-10-31:
Interesting. Where did the exception occur?

I repeated the same test sequence using both QGC and MP as the GCS, and saw no problems.
For MP, I started SITL this way (I had never used MP with SITL before) and connected to the default TCP port:
```
(base) 08:11:02 (pr-ekf-init-fail) ~/linux_git/kd0aij/ardupilot/ArduCopter/sitl$ ../../Tools/autotest/sim_vehicle.py -D -G --no-mavproxy
```
I also flashed the firmware to an mRoX21 and tested using MP as the GCS.  No problems there either.

kd0aij, 2019-11-01:
There is a bug in master which results in a segfault in SITL if EK2_IMU_MASK is zero.
I am able to reproduce that in master, but it is fixed in #12710.

rmackay9, 2019-11-26:
I've retested with master now that the above linked PR has gone in and this issue is resolved, thanks!

closed 2019-11-26 by rmackay9