ArduPilot/ardupilot #12715: Compile error in Copter 4.0 if I disable optical flow

andyp1per, 2019-10-30:
I set
```#define OPTFLOW               DISABLED  ```
in APM_Config.h and got this:

[551/560] Compiling ArduCopter/mode_guided_nogps.cpp
../../ArduCopter/ekf_check.cpp: In member function 'bool Copter::ekf_over_threshold()':
../../ArduCopter/ekf_check.cpp:113:10: error: 'optflow' was not declared in this scope
     if (!optflow.healthy() && (vel_variance >= (2.0f * g.fs_ekf_thresh))) {
          ^
compilation terminated due to -Wfatal-errors.

Waf: Leaving directory `/Users/andypiper/Documents/GitHub/ardupilot-released/build/KakuteF7Mini'

### Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2019-10-30:
Unassigned myself 'cuz there's lots more people here than just me

peterbarker, 2020-01-07:
I fixed this a while back....

closed 2020-01-07 by peterbarker