ArduPilot/ardupilot #12718: Issues with running SITL

emd802, 2019-10-30:
### Bug report

**Issue details*

When I try to run sim_vehicle.py --map --console to start up the SITL, I get the following message on my command prompt:  

SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
[pkill] An exception has occurred with command: 'pkill ArduSub.elf'
[Error 2] The system cannot find the file specified
SIM_VEHICLE: Killing tasks
[pkill] An exception has occurred with command: 'pkill ArduSub.elf'
[Error 2] The system cannot find the file specified
Exception KeyError: KeyError(36468,) in <module 'threading' from 'C:\Python27\lib\threading.pyc'> ignored

When running it on my Cygwin, I get the following message:
$ python sim_vehicle.py --map --console
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
Failed to find location (CMAC)
SIM_VEHICLE: Killing tasks

I'm trying to do this for a copter set up, and it worked great before on my previous box, but for some reason I can't get it to work on this one.

If someone has some info that will help PLEASE help me out with this.

peterbarker, 2019-10-31:
On Wed, 30 Oct 2019, emd802 wrote:

> [pkill] An exception has occurred with command: 'pkill ArduSub.elf'

I think you may need to install the procps cygwin package.

> Failed to find location (CMAC)

That one's just weird.  I think you'd need to add debug to trace that one 
further.

peterbarker, 2020-01-07:
@emd802 any update?

azhdarshirinzada, 2020-01-16:
it seems the same:

../Tools/autotest/sim_vehicle.py --map --console
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
[pkill] An exception has occurred with command: 'pkill ArduSub.elf'
[Errno 10] No child processes
SIM_VEHICLE: Killing tasks
[pkill] An exception has occurred with command: 'pkill ArduSub.elf'
[Errno 10] No child processes

jhn002, 2020-04-07:
Hello hello,
@azhdarshirinzada , @emd802 does anyone have some work around to this issue? Is switching to a virtual machine the easiest thing to do?

azhdarshirinzada, 2020-04-07:
@jhn002 I solved this issue by changing the branch to Copter3.6

jhn002, 2020-04-07:
Thanks @azhdarshirinzada, I'll try that out before giving into to vm.

peterbarker, 2020-04-08:
@khancyr Do you have a cygwin setup you can find this regression with?

khancyr, 2020-04-08:
Hello,
No I don't have cygwin setup. But it shouldn't be trying to kill a .elf file, we don't have extension anymore on our SITL files

safashaikh, 2020-04-08:
I had to modify the sim_vehicle.py lines 169-171 to be:
`def under_cygwin(): `
   `     """Return if Cygwin binary exist""" `
    `    return os.path.exists("C:\\cygwin64\\bin\\cygstart.exe")`