ArduPilot/ardupilot #12720: Desc field in Mission Planner not wrappting

HilldaFlyer, 2019-10-30:
### Bug report

**Issue details**
In the full parameter tab, changing values opens a new dialog box, The Desc message doesn't wrap so it cannot be read.
_Please describe the problem_

**Mission Planner 1.3.68 build 1.3.78105.24678 **
_What version was the issue encountered with_
ArduPlane v4.0.0
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Plane
**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
PixRacer
**Logs**
_Please provide a link to any relevant logs that show the issue_

peterbarker, 2020-01-07:
@HilldaFlyer do you think this should be an issue on the MissionPlanner repository rather than the ArduPilot one?

rmackay9, 2020-01-07:
Here's a link to the MP issues list: https://github.com/ArduPilot/MissionPlanner/issues

closed 2020-01-07 by rmackay9