ArduPilot/ardupilot #12722: Copter: GCS Failsafe should flash LED yellow

rmackay9, 2019-10-31:
During testing PR https://github.com/ArduPilot/ardupilot/pull/7432 which greatly improves the GCS failsafe I found that the LED was remaining Green which made it slightly less obvious to the pilot why the vehicle was entering RTL.

I'm not sure if we should also play the buzzer... maybe not..

Pedals2Paddles, 2019-10-31:
Given my love for lights and buzzers, I'm surprised I didn't think of this when doing the failsafe PR.  I will add this to the failsafe PR this evening.

peterbarker, 2020-01-07:
@Pedals2Paddles if you did - I don't think you pushed it?

Pedals2Paddles, 2020-01-07:
After discussing the implications of lights, buzzers, and pre-arm checks with Randy, we decided not to do those in the original PR.  But this should still be done soon.  I'll get back to this one soon.

Pedals2Paddles, 2020-01-07:
Ha, I forgot I did a PR for this two months ago when I pulled it out of the refactor.  Just rebased it. https://github.com/ArduPilot/ardupilot/pull/12790

rmackay9, 2020-02-05:
PR merged so closing, thanks!

closed 2020-02-05 by rmackay9