ArduPilot/ardupilot #12723: Copter: add pre-arm check of GCS failsafe

rmackay9, 2019-10-31:
I think we should add a pre-arm check of whether the GCS failsafe has triggered.  We have similar pre-arm checks for RC, battery and ekf failsafes.

TBH, it's possible this may be more annoying than it is useful but without it there is some danger that the user will have triggered the GCS failsafe (perhaps the radio disconnects before takeoff) but the user doesn't notice and they end up flying without an active GCS failsafe.

Pedals2Paddles, 2019-10-31:
IMO this definitely should be a pre-arm check. It can easily be disabled with the arming check bitmask for operators that don't want/need it. That being said, I think you're right in that it will certainly annoy people at first.  I'm ok with leaving this out of the failsafe PR so as not to completely torture users with multiple behavior changes at once :)

Pedals2Paddles, 2020-01-07:
Since the default GCS Failsafe is now disabled, I think this should now be a pre-arm check that won't cause user heartburn. It won't do anything at all if no GCS has ever been connected, which is also good.  I'll get on this shortly.

Pedals2Paddles, 2020-02-06:
Closed by https://github.com/ArduPilot/ardupilot/pull/12790

closed 2020-02-06 by Pedals2Paddles