ArduPilot/ardupilot #12750: ADSB/Avoidance Issues: Plane enable, incorrect nomenclature

Hwurzburg, 2019-11-02:
@peterbarker  @tridge @rmackay9 ...I am trying to understand a few ADSB/Object Avoidance things in order to better document, as well as make sure there are desireable user controls on its enabling by RC option switches...I have spoken to Tridge and Randy some, but am afraid that I am still confused

my current understanding:
-there are basically two types of avoidance triggers: ADSB and Rangefinder
-once triggered, until the threat is passed, the current flight mode  is changed (change allowed only with limitations) to a guided one to avoid, then returned to previous mode after threat is passed

The enabling params are:
ADSB_ENABLE param: enables collection of targets via adsb receiver
AVD_ENABLE param: enables/disables entry into the avoidance mode (I think currently mis-labeled as AVOID_ADSB in modes, since I believe it forces the guided avoidance mode for ADSB or Object, ie Rangefinders), and is incorrectly described in flight mode params

vehicle.adsb.enable() returns ADSB_ENABLE param value
vehicle.avoidance.enable() returns value of AVD_ENABLE param
vehicle.avoidance_adsb.enable() enables/disables entry into "AVOID_ADSB" modes

if my understanding is correct, I need to do the following:
1. correctly describe the AVOID_ADSB mode in the wiki for the vehicles and I understand we can't correct the name without screwing up something (is this correct?)
2. change the name of the Copter RC option to Avoidance enable/disable and describe it as pertaining to both ADSB and Object avoidance
3. Add a mavlink mesage to RC option change to say Avoidance Enabled, Disabled, or "No avoidance mode has been set" with switch changes, for user notification like some other RC option switches have...I think its important the user knows he disabled or enabled such a significant behavior changer..
4. Add an Avoidance enable/disable to Plane for an RC option, since none exists currently...

if the above is correct, I will create the PRs in code and wiki, otherwise, please educate me....thanks

**Platform**
[x  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2019-11-02:
On Sat, 2 Nov 2019, Henry Wurzburg wrote:

> my current understanding:
> -there are basically two types of avoidance triggers: ADSB and Rangefinder

Actually, there are essentially two entirely separate avoidance schemes 
used in ArduPilot, traditionally "macro" and "micro".

ADSB is the only real trigger for macro avoidance, 'though you can also 
trigger based on mavlink vehicle position messages IIRC.  We'll probably 
also add AIS in for this.  Rangefinders are *unused* in "macro" avoidance 
mode.

Micro avoidance takes data from a lot of places - Proximity sensors, 
fences, rangefinders, ....

Plane only does "macro" avoidance at this point.  Copter, Rover and Sub 
can all do micro-avoidance too.  Tracker does neither :-)

> -once triggered, until the threat is passed, the current flight mode is changed (change allowed only with limitations) to a guided one to avoid, then
> returned to previous mode after threat is passed

This is true of the "macro" avoidance stuff.  "micro" avoidance just tries 
to steer around things in the current mode.

> The enabling params are:
> ADSB_ENABLE param: enables collection of targets via adsb receiver
> AVD_ENABLE param: enables/disables entry into the avoidance mode (I think currently mis-labeled as [...]

AvoidADSB is a mode used by Copter to handle the macro avoidance.  Plane 
may get one eventually, how it's been restructired.

> if the above is correct, I will create the PRs in code and wiki, otherwise, please educate me....thanks

I think we might need to clear up the distinction between the two 
avoidance regimes before we attach your other issues in here. 
Additionally, I'm not sure if we've now added code into the 
micro-avoidance stuff for ADSB vehicles (via the OA Database stuff). 
@MagicRub?

Hwurzburg, 2019-11-02:
I think Plane already has the AVOID_ADSB mode....

IamPete1, 2021-08-15:
@Hwurzburg is this resolved?

peterbarker, 2024-11-24:
Ping @Hwurzburg - NeedAnswer

closed 2024-12-02 by Hwurzburg