ArduPilot/ardupilot #12753: Rover Ignores DO_CHANGE_SPEED

jimovonz, 2019-11-03:
### Bug report

**Issue details**
Rover Ignores DO_CHANGE_SPEED command during mission. Have tried on vehicle and SITL with same response. Works fine in Rover 3.5.

**Version**
Rover 4.0 + master (fmuv3)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

**Airframe type**
Skid steer

**Hardware type**
Pixhawk Cube Black

jimovonz, 2019-11-03:
It seems that there is some confusion in the code around the scope of the variable  _desired_speed.

Defined here:
https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/mode.h#L210
distinct from the private class variable defined here:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AR_WPNav/AR_WPNav.h#L142

_desired_speed as used in the APMRover2/. files is not used in navigation.

A quick fix for this issue is to insert the following after this line:
https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/mode.cpp#L243

        g2.wp_nav.set_desired_speed(speed);

rmackay9, 2019-11-04:
@jimovonz, thanks for the report and the investigation.  I've also reproduced the issue and I agree that the confusion between the internal variable and the navigation library's equivalent is the root of the problem.

magicrub, 2019-11-04:
I've already fixed this on a branch I have somewhere. I'll see about making a PR for it.

rmackay9, 2019-11-04:
@magicrub, before you put too much time into digging it out, I've done a relatively extensive clean-up of the _desired_speed handling by moving this variable down into each flight mode.  https://github.com/rmackay9/rmackay9-ardupilot/commits/rover-change-speed-fix

I suspect there were some other places where we would get odd behaviour.

I'll test this a bit more and then create a PR.

rmackay9, 2019-11-05:
I've created a PR which resolves this issue: https://github.com/ArduPilot/ardupilot/pull/12772

Any feedback is greatly welcome!

rmackay9, 2019-11-08:
Fixed in master and included in Rover-4.0.0-rc3.  Thanks very much @jimovonz!

closed 2019-11-08 by rmackay9