ArduPilot/ardupilot #12762: Arduplane: bad CRC on mavlink packets

raul-ortega, 2019-11-04:
# Bug report

I'm not sure whether it is a bug or not, and whether it is caused by Ardupilot or Mavlink library, and I apologize if this has been treated before but my search were unsuccessful.

**Issue details**

When parsing mavlink messages several frames are droped by MAVLINK_FRAMING_BAD_CRC condition. The bad CRC are repited many times and always when specifics message ids arrives: 152, 178, 182, 163, 165, 168, 193 ... (order and number of repetition might vary on time).

I parsed the frames with mavlink c_library_v1, captured from Arduplane 3.9.7 (Chibios) on a quadplane (tricopter tilt vectored yaw). I also parsed data captured by two friends, with similar setup.

**Version**

Arduplane 3.9.7
mavilin c_library_v1

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

Quadplane (tricopter tilt vectored yaw).

**Hardware type**

Pixhack 2.8.4

**Logs**

Data captured from telem port:

[capture1](https://drive.google.com/open?id=143K9Gh4hXc4otwEHDjjjhiwRS6FU82gK)
[capture2](https://drive.google.com/open?id=1y-tZbatS0zasPD0xPoP_TvAIdPjKF5i-)
[capture3](https://drive.google.com/open?id=1EYEJMyo2KnFgk7WTzZ4s6dqNdT7gfJ3o)

raul-ortega, 2019-11-04:
Some comments:

capture1 is done through Dragon Link RX/TX radio system.

capture2 is done through TBS crossfire (transparent mavlink).

capture3 is done thorugh direct connection to telem2 port with FTDI

SRx params are different on all captures.

raul-ortega, 2019-11-07:
After upgrading to c_library_v2 the issue persists

raul-ortega, 2019-11-07:
This is the source code of the parser I'm using:

#include <iostream>
#include <sstream>
#include <fstream>
#include <cstdio>
#include <cmath>
#include "mavlink/common/mavlink.h"

#define SIZE 40960

void mavlink_encodeTargetData(uint8_t c);
uint8_t percent(uint32_t A, uint32_t B);

uint16_t telemetry_failed_cs = 0;
mavlink_status_t status;

using namespace std;

int main(int argc,char * argv[])
{
    size_t buffer_size = SIZE;
    char *buffer = new char[buffer_size];

    std::ifstream fin(argv[1]);

    printf("CRC,ID,SEQ,LEN\n");

    while (fin)
    {

        fin.read(buffer, buffer_size);

        size_t count = fin.gcount();

        if (!count)
            break;

        for(uint16_t i=0;i<count;i++){
            mavlink_encodeTargetData(buffer[i]);
        }
    }
    cout << endl;
    delete[] buffer;

    printf("# SUMMARY #\n");
    printf("TOTAL:  %d\n",status.packet_rx_success_count + telemetry_failed_cs);
    printf("GOOD:   %d (%d%%)\n",status.packet_rx_success_count, percent(status.packet_rx_success_count, status.packet_rx_success_count  + telemetry_failed_cs));
    printf("BAD:    %d (%d%%)\n",telemetry_failed_cs,percent(telemetry_failed_cs, status.packet_rx_success_count + telemetry_failed_cs));

}

void mavlink_encodeTargetData(uint8_t c) {
    mavlink_message_t msg;
    if (mavlink_parse_char(0, c, &msg, &status)){
        printf("GOOD,%d,%d,%d\n",msg.msgid,msg.seq,msg.len);
        //mavlink_handleMessage(&msg);
    } else if(status.packet_rx_drop_count > 0){
        printf("BAD,%d,%d,%d\n",msg.msgid,msg.seq,msg.len);
        telemetry_failed_cs += status.packet_rx_drop_count;
    }
}

uint8_t percent(uint32_t A, uint32_t B){
    return round(((float)A * 100) / (float)B);
}

raul-ortega, 2019-11-07:
The issue and pull request bellow might be related:

[Issue #237](https://github.com/ArduPilot/pymavlink/issues/237)
[PR #](https://github.com/mavlink/mavlink-devguide/pull/135)

peterbarker, 2024-07-31:
Gah.  Sorry, this looked like a good report which we never acted upon.

Almost 5 years on - did you want to pursue this, @raul-ortega  ?

peterbarker, 2024-09-04:
OK, no response and probably too old to reasonably chase.

Please create a fresh issue if this is reproducible on master.

closed 2024-11-24 by peterbarker