ArduPilot/ardupilot #12764: Plane: Possible dangerous behavior with Vtol_Takeoff

EERomeo, 2019-11-04:
### Bug report

**Issue details**

I was doing an auto mission today and I programmed the mission with Vtol_takeoff since the plane is a vtol. Upon arriving at the location, due to wind and planned takeoff location change, I decided to perform the takeoff procedure myself.
All was well, liftoff, than transition to FBWA, climb to about 230m, than changed mode to Auto. At this point it looks like the plane wanted to take it from beginning, including Vtol_takeoff, so it switched to Q mode and wanted to do the takeoff procedure again, while at 230m altitude. 
I was watching it so while I did not understand at that time what is going on, I switched to FBWA again saving the day. After this all went well, the second time switching to Auto it proceeded in plane mode to do it's programmed mission.

Now, I know the Nav_Takeoff in normal plane works as it should, if the plane is already flying, this command is ignored and next waypoint is executed. It seems that Nav_Vtol_Takeoff is not the same.

**Version**
Plane 3.9

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Vtol tilt rotor

**Hardware type**
CUAV Pixhack V3

**Logs**
https://drive.google.com/open?id=10Ow6YMDt3bJC6Yf_N5UFyfUVzxZi7kZA

WickedShell, 2019-11-04:
@EERomeo I agree strongly with this. I couldn't get @tridge to change the default behavior to avoid this, but it's extremely surprising when this happens. I did actually put in code to avoid this, you can set the bit in `Q_OPTIONS` for respect takeoff frames to fix this for now. http://ardupilot.org/plane/docs/parameters.html#q-options-quadplane-options

closed 2019-11-04 by WickedShell

EERomeo, 2019-11-04:
Wow, didn't know this was discussed before.
Thanks for the fix, this works for me, although I do not see a scenario where this should not be the default behavior. Or at least a hint in the wiki.