ArduPilot/ardupilot #12781: AP_Logger: problems with block based flash memory

anbello, 2019-11-06:
### Bug report

**Issue details**

When logging on FC with only flash memory (Revo mini, omnibus nano, kakute F7, ...) frequently we have totally or partially missing logs, glitching logs. Whith clean flash and First Flight After boot Is more frequent to have a good log.

Some Report from users:
https://discuss.ardupilot.org/t/tuning-the-harmonic-notch/48836/8
https://discuss.ardupilot.org/t/tuning-the-harmonic-notch/48836/10

**Version**
4.0

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

shellixyz, 2019-11-14:
#12483 will probably fix this issue

anbello, 2019-11-15:
Yes I can confirm what @andyp1per  wrote. There are block based logging problems also with 4.0rc1 and 4.0rc2.

shellixyz, 2019-11-15:
@andyp1per @anbello Maybe I made a mistake but I couldn't find the related commit in the ArduPlane-4.0.0 tag

andyp1per, 2020-01-02:
I think I have fixed all the common issues people encounter in https://github.com/ArduPilot/ardupilot/pull/13130 - I have also added some code that copes with finding corruption in the flash. Hence closing. Let me know if anything looks like it is still not working.

closed 2020-01-02 by andyp1per