ArduPilot/ardupilot #12782: Could you extend the frequency range for parameters SR0_xxxx (Mavlink stream rates)?

FelipeSanchezCalzada, 2019-11-06:
### Feature request

**Is your feature request related to a problem? Please describe.**
Why are the values ​​of SR0_ Parameters limited to 10Hz?
I would like to be able to use 30Hz in several projects I am working on.
Is it a limitation caused by the processor or memory?

**Describe the solution you'd like**
Simply increase the 10Hz limit to that group of parameters. For example up to 25Hz or 30Hz

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Thanks to all the developers for this project, regards!

IamPete1, 2019-11-07:
Have you tried setting 30Hz? I don't think they are hard limits, just recommended ranges.

Georacer, 2019-11-07:
I have the same question and I add to it: Given that the I/O MAVLink
channels of the Pixhawk are pretty busy, can we expect some guaranteed
output rate when setting high requested rates?

For example, the ROSFlight project does declare that the default MAVLink
output rates for attitude are 100 Hz and can go up to 1000Hz (IRRC).

Thanks.

IamPete1, 2019-11-07:
> Given that the I/O MAVLink channels of the Pixhawk are pretty busy, can we expect some guaranteed output rate when setting high requested rates?

I don't think we can guarantee any output rate, but with faster processors ect we should be able to do it quite fast so long as your baud rate is high enough to get all the data out. 

@peterbarker is one of the experts on this stuff

peterbarker, 2019-11-07:
On Thu, 7 Nov 2019, Peter Hall wrote:

> I don't think we can guarantee any output rate, but with faster processors ect we should be able to do it quite fast so long as your baud rate is high
> enough to get all the data out.

Messages are limited to 80% of the main loop rate.  So 320Hz on Copter's 
main loop rate.

Not many messages will have interesting data in them at that rate.

These can be requested using the using the mavlink message 
SET_MESSAGE_INTERVAL.

Georacer, 2019-11-08:
Personally, my interest is regarding off-board attitude control, thus the need for higher rates.
My experience (from reading what other people say) is that Ardupilot boards may not be the best way to go about this.
This is why recently I've been trying out this project: https://rosflight.org/
More bare-bones feature-wise, but with more mature off-board control (hopefully).

rmackay9, 2019-11-09:
I think ardupilot probably has all the control you'll need especially now that we've added support for the set-message-interval stuff.  I think it's also very reliable so you'll spend less time trying to get your vehicle to fly (and not crash) and more time on the part that's important for you.

What we don't have enough of is documentation.  In particular we need to improve this part of the wiki: http://ardupilot.org/dev/docs/mavlink-commands.html

FelipeSanchezCalzada, 2019-11-09:
Sorry for the delay, I've been busy these days.

https://github.com/ArduPilot/ardupilot/issues/12782#issuecomment-550572114
Right now I have tried to set values ​​much higher than 10 Hz, and it works! 
I really thought it was a limitation.

https://github.com/ArduPilot/ardupilot/issues/12782#issuecomment-551603201
Although it is outside my question I found this reference to RosFlight very useful. Thanks!

https://github.com/ArduPilot/ardupilot/issues/12782#issuecomment-552044834
I totally agree with you. Ardupilot is great but its documentation sometimes becomes confusing in some aspects. It is logical due to the size of the project.

closed 2019-11-09 by FelipeSanchezCalzada