ArduPilot/ardupilot #12792: AP_OSD: Save screen space, more options

SlipstreamFPV, 2019-11-08:
Compared with a minimOSD running night-ghost firmware the OSD is lacking some features.

I would like to see the following options:

Option to remove the leading/trailing icons on items. e.g. "lat/lon" that preceeds the gps co-ords or "GS" and the arrow that preceeds the ground speed.

On night-ghost firmware the element can be turned on and off just like the ardupilot OSD but then there are sub options like "Show arrow" and "Show icon before value"

There are also options for the GPS such as "use less precision" that lowers the decimal numbers to 5 instead of the standard 6 on night-ghost firmware. Ardupilot OSD uses 7 which is nice if you need that precision but it's overkill for a lot of users and uses up extra screen real-estate.

5 decimal places is acurate to 1.1m already, 7 is acurate to 11mm, which most of our GPS units aren't even capable of so an option here would be amazing!

Finally with GPS they also have the option to have only the decimal numbers show. This is excellent for sub screens to declutter even further while still giving the information that's needed if the plane goes down. The whole numbers are easy to work out, they don't need to be displayed all the time in a lot of circumstances. 

Currently 'lat' on the GPS occupies 12 characters, this could be as little as 5 with the options above.

Platform
[X] All

Additional context
Thanks for taking the time to consider these enhancements!

SlipstreamFPV, 2019-12-03:
I guess this is too hard or no one cares?

auturgy, 2019-12-03:
It might just be that the right people haven’t seen it: a lot of the OSD features have come from community contributors, not core dev team members (who get GitHub notifications).
Perhaps have a look in the forum for a relevant thread, and tag this into it?
Just a thought.

Regards,

James