ArduPilot/ardupilot #12797: Plane: SDP3X causes continuous watchdog resets

Naterater, 2019-11-08:
#### Issue details
Unfortunately upgrading to the stable release of plane causes continuous rebooting whenever the SDP3X airspeed sensor is plugged in. It might last 3 or 4 minutes before the first reboot, but thereafter it's usually less than 30 seconds between reboots. When I tested the 4.0.0 beta2 release, everything appeared OK.

#### Version
Plane 4.0.0 stable... NOT Plane 4.0.0 beta 2.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Twin Electric Elanus Duo

#### Hardware type
Cube Orange, also confirmed on Cube Blue(black). I had an apj for the cube black of 4.0.0 beta2, and when I tried that, it worked. Didn't have an .apj of beta 2 for the cube orange, so couldn't try that.

Drotek SDP3X airspeed sensor, Here2 on serial, RFD900+ at 20dB on telem1, ESC telemetry to serial 4, camera and servos over main and aux outs.

#### Logs
https://1drv.ms/u/s!AunBK4rE1ZDCgcgjiW712ywQCz52uw?e=0mcXDc

rmackay9, 2019-11-08:
I had a quick look at the logs and it looks like the watchdog is jumping to to restart the board.  I think this means it's unlikely to be a brown-out and more likely to be a software issue.
[image]

peterbarker, 2019-11-10:
Two possible fixes:
 - https://github.com/ArduPilot/ardupilot/pull/12807/files
 - https://github.com/ArduPilot/ardupilot/pull/12808/files

Naterater, 2019-11-11:
Everything works on master when I tested a cube orange build moments ago. Will close the issue as soon as it makes it to a stable release.

WickedShell, 2019-12-07:
Shipped in 4.0.1

closed 2019-12-07 by WickedShell