ArduPilot/ardupilot #12798: AP_Scripting/AP_Arming: add a pre-arm check that scripting has initialised

rmackay9, 2019-11-08:
I've found that if scripting fails to initialise because the SCR_HEAP_SIZE was set to be too large for the board's available memory (i.e. SCR_HEAP_SIZE = 90k on a CubeBlack) the scripting fails to initialise but there is no visible warning to the user.

We should add a pre-arm check (perhaps to the systems category?) that scripting has failed to initialise.

I tested this by creating a very simple script that used send_text to send, "Hello World" to the GCS.  This ran find when SCR_HEAP_SIZE = 60k but failed once raised to 90k

IamPete1, 2019-11-08:
would be nice to have a failsafe action if scripts crash for some reason also, that might be trickery to implement though

rmackay9, 2019-12-03:
Resolved in master and this will go out with Copter-4.0.0-rc3

closed 2019-12-03 by rmackay9