ArduPilot/ardupilot #12805: RC inoperable if RCx_OPTION set to invalid value

rmackay9, 2019-11-09:
During Rover-4.0.0 testing a user has found that if they have an RCx_OPTION param set to an invalid value (like "1") then two bad things happen:

- RC inputs no longer work at all for any channel
- "Config error: Failed to init: RCx_OPTION: 1" is streamed to the GCS every few seconds

Even if the user changes the RCx_OPTION to a reasonable value the RC input does not recover nor does the error message stop.  This is a unnecessarily extreme response to a small configuration issue.

kd0aij, 2019-11-09:
The configuration error is currently detected at boot, and the firmware enters an error loop which emits the error message every 3 seconds. To correct the problem you must first set the parameter to a valid value, then reboot. This is the reason no RC channels (or anything else) appear to be working.

Prior to this change there was a single error message emitted at bootup, and again in flight when you  attempted to execute the function. The new behavior is intended to make it obvious at boot-time that something is wrong, while providing a convenient way to fix the problem.

Hwurzburg, 2019-11-10:
what does it do in flight?

kd0aij, 2019-11-10:
You can't do anything until you fix the parameter and reboot.
I suppose the error loop could be modified to treat an invalid RC option differently from other configuration errors in order to restore behavior to ignoring the problem.

Hwurzburg, 2019-11-10:
So change to invalid option in flight= crash??!

kd0aij, 2019-11-10:
No, the only effect _should_ be to generate a console message complaining "...invalid option..." assuming that is the only problem...
But if there's an invalid RC option set, maybe other things are screwed up, such as might happen if parameters for the wrong vehicle were loaded.

Hwurzburg, 2019-11-10:
good, just checking...

rmackay9, 2019-11-10:
@kd0aij,

This is really not good at all.  This changes an innocuous config error into a real problem.  We should revert this change or fix it.

kd0aij, 2019-11-11:
I'm not sure whether requiring the parameter be set to a valid value and rebooting before flight constitutes a "real problem", or "better error reporting" :)

But perhaps I misunderstood the conclusion on the devcall, and this was only supposed to apply to SITL builds.,,  Would that be an acceptable fix?

rmackay9, 2019-11-11:
@kd0aij, I think better warning would have been ok but this actually led to subsystems (like RC) not working at all.  Anyway, thanks for the review on the [new PR](https://github.com/ArduPilot/ardupilot/pull/12811).

rmackay9, 2019-11-21:
This has been fixed in master and rolled out with Rover-4.0.0-rc4 so closing, thanks!

closed 2019-11-21 by rmackay9