ArduPilot/ardupilot #12809: Vagrant times out while booting using Virtualbox 6

chigozienri, 2019-11-10:
### Bug report

**Issue details**
When running "vagrant up", the virtual machine boots but vagrant times out in the console after `bionic64: SSH auth method: private key`. This does not happen when I downgrade to virtualbox 5.2

It seems to be the same issue as here: https://github.com/hashicorp/vagrant/issues/10578

**Version**
OS X 10.14, Vagrant 2.2.6, VirtualBox 6.0.14

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
n/a

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
n/a

**Logs**
_Please provide a link to any relevant logs that show the issue_
Console output on running "vagrant up" with virtualbox 6:
```
➜ ardupilot git:(master) vagrant up
Bringing machine 'bionic64' up with 'virtualbox' provider...
==> bionic64: Checking if box 'ubuntu/disco64' version '20191107.0.0' is up to date...
==> bionic64: Clearing any previously set forwarded ports...
==> bionic64: Clearing any previously set network interfaces...
==> bionic64: Preparing network interfaces based on configuration...
    bionic64: Adapter 1: nat
==> bionic64: Forwarding ports...
    bionic64: 22 (guest) => 2222 (host) (adapter 1)
==> bionic64: Running 'pre-boot' VM customizations...
==> bionic64: Booting VM...
==> bionic64: Waiting for machine to boot. This may take a few minutes...
    bionic64: SSH address: 127.0.0.1:2222
    bionic64: SSH username: vagrant
    bionic64: SSH auth method: private key
Timed out while waiting for the machine to boot. This means that
Vagrant was unable to communicate with the guest machine within
the configured ("config.vm.boot_timeout" value) time period.

If you look above, you should be able to see the error(s) that
Vagrant had when attempting to connect to the machine. These errors
are usually good hints as to what may be wrong.

If you're using a custom box, make sure that networking is properly
working and you're able to connect to the machine. It is a common
problem that networking isn't setup properly in these boxes.
Verify that authentication configurations are also setup properly,
as well.

If the box appears to be booting properly, you may want to increase
the timeout ("config.vm.boot_timeout") value.
```

peterbarker, 2019-11-10:
Wow.  That's nasty.

We do need the serial stuff - allows people to flash hardware from the VM.

Perhaps we should offer the bento boxes as options in Vagrantfile?

chigozienri, 2019-11-11:
That seems to be Vagrant's recommendation anyway, at the end of this page https://www.vagrantup.com/docs/boxes.html

> It is often a point of confusion, but Canonical (the company that makes the Ubuntu operating system) publishes boxes under the "ubuntu" namespace on Vagrant Cloud. These boxes only support Virtualbox and do not provide an ideal experience for most users. If you encounter issues with these boxes, please try the Bento boxes instead.

I have been playing around with using bento/ubuntu-19.10 this morning. It doesn't work with the initvagrant.sh file as currently written. During initial provisioning it halts, giving this error:
```
bento: + sudo resize2fs /dev/sda1
    bento: resize2fs 1.45.3 (14-Jul-2019)
    bento: Couldn't find valid filesystem superblock.
    bento: resize2fs: Device or resource busy while trying to open /dev/sda1
```
I removed the `sudo resize2fs /dev/sda1` line from initvagrant.sh and successfully built the vagrant environment and launched the arducopter simulation using Virtualbox 6.

There are 2 unrelated issues I discovered while trying to edit the Vagrantfile to add a bento box option, which complicate writing a new Vagrantfile with this fix and making a pull request:

Issue 1:
In my log in the OP, observe that line 3 says `bionic64: Checking if box 'ubuntu/disco64' version '20191107.0.0' is up to date...` - bionic64 is marked as the primary box in the Vagrantfile, but the line `config.vm.box = "ubuntu/disco64"` at line 141 seems to override this, even though it should only do so when you specify the disco64 box (using `vagrant up disco64`).

Issue 2:
I made a Bento option in the Vagrantfile as follows:
```
  config.vm.define "bento", autostart: false do |bento|
    config.vm.box = "bento/ubuntu-19.10"
    config.vm.provision :shell, path: "Tools/vagrant/initvagrant-bento.sh"
    config.vm.provider "virtualbox" do |vb|
      vb.name = "ArduPilot (bento)"
      vb.gui = false
    end
  end
```
where initvagrant-bento.sh is the same as initvagrant.sh with the `sudo resize2fs /dev/sda1` line removed. (sidenote: I set vb.gui = false because the bento box is CLI only by default anyway, but it did work with vb.gui = true)
However, when I brought it up using `vagrant up bento` it would only provision using the original initvagrant.sh file (and therefore fail). This persisted until I removed every other box option from the Vagrantfile. I suspect that the first `config.vm.provision :shell, path: "Tools/vagrant/initvagrant.sh"` in the file is overriding all the others for some reason.

It might be that I am just misunderstanding the Vagrantfile format. If not, I could open a separate bug report for these 2 issues.

peterbarker, 2019-11-11:
On Mon, 11 Nov 2019, chigozienri wrote:

> However, when I brought it up using vagrant up bento it would only provision using the original initvagrant.sh file (and therefore fail). This persisted
> until I removed every other box option from [...]

OK, that's weird.  Happily, the fix for that is also a good change to make 
in your methodology; adjust the existing script to cope with running on 
the bento box.  If you insert "set" into the initvagrant script and 
inspect the output you ought to be able to spot something in the 
environment you can differentiate the bento boxes with.  It's also 
possible the output of "lsb_release" could help there.

If it's just one problematic line this should work reasonably well.

> It might be that I am just misunderstanding the Vagrantfile format. If not, I could open a separate bug report for these 2 issues.

Well, I thought the nominated scripts were per-VM...

Thanks,

Peter

IamPete1, 2021-08-15:
sounds like this was resolved.

closed 2021-08-15 by IamPete1