ArduPilot/ardupilot #12815: Copter: LandingGear: RETRACT displayed when no landing is setup

rmackay9, 2019-11-11:
I've noticed in SITL that the "LandingGear: RETRACT" message appears even if there are no landing gear setup.  This will lead to unnecessary support calls from users so we should remove this message.

We should either remove the message completely or ensure that it's only sent if a SERVOx_OUTPUT has been set to LandingGear.  The [code is here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_LandingGear/AP_LandingGear.cpp#L142).

R-Lefebvre, 2019-11-12:
What would invoke this message to appear Randy?  I have not seen it yet.

rmackay9, 2019-11-12:
@R-Lefebvre, it was appearing for me after a takeoff mission command completed... it's possible that a lidar also had to be fitted to the vehicle to make it appear but I don't think so.

This has been fixed in master now though and will go out with Copter-4.0.0-rc3 so I'll close it I think

closed 2019-11-12 by rmackay9