ArduPilot/ardupilot #12818: Copter: Auto can become stuck on takeoff command if altitude error is large (?)

rmackay9, 2019-11-11:
An auto mode may become stuck on the takeoff command if these steps are performed in SITL:

- enable the rangefinder with this parameter file: https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/default_params/copter-rangefinder.parm.  I.e. param load ../Tools/autotest/default_params/copter-rangefinder.parm and then reboot SITL
- upload a mission like this:
   - takeoff to 5m
   - waypoint to some point (does not matter where)
   - return-to-launch (probably also not relevant)
- arm the vehicle in Loiter
- rc 3 2000 (a rapid takeoff seems to be necessary)
- switch to auto when the vehicle reaches about 10m

It seems the vehicle is stuck on the takeoff command (I added some extra logging in the 1hz loop to see which command it was on).  It seems like AC_WPNav's reached_destination() is perhaps not true.
[image]

rmackay9, 2019-11-12:
This is fixed in master now and will go out with Copter-4.0.0-rc3 so closing

khancyr, 2019-11-12:
I would like to keep it open until we can fully understand what the root cause of the failure. 
As discussed with @peterbarker the same stucking append on Heli, I also saw it on precland (modified to use wp_nav on first part).  So we get a was to stuck the wp_nav controler and that is not good...

khancyr, 2019-11-12:
Could be related to https://github.com/ArduPilot/ardupilot/pull/11387/files (that I never manage to reliably prove, and aren't confident on the fix)

rmackay9, 2019-11-13:
@khancyr,  Thanks for all the input on this issue.

I'm going to go ahead an close this issue (again).  The reason is that I raised this issue for a very specific and real issue that I discovered.  If we leave it open as-is I fear it will just sit there forever with no resolution.

I agree that there is potentially another bug but it's not clear that it can ever actually happen.  Please feel free to raise a new issue if you like though, I won't object.

closed 2019-11-13 by rmackay9