ArduPilot/ardupilot #12823: Copter: SBus.out on Pixhawk Cube 2.1 doesn't work

jasin755, 2019-11-11:
I have tried to connect something to S.Bus out of Pixhawk Cube and it never haven't worked.  I have tried to connect logic analyzer and always no signal appears. I have tried to set `BRD_SBUS_OUT` to `1` and nothing changed. Where can be problem? 
Thanks.

rmackay9, 2019-11-13:
This is really a support issue so it's best if it's first raised in the Copter-3.6 or Copter-4.0 discussion threads.  Sorry to be strict but if we're not then our to-do list turns into another support forum.

Copter-3.6 category: https://discuss.ardupilot.org/c/arducopter/copter-36
Copter-4.0 category: https://discuss.ardupilot.org/c/arducopter/copter-40

closed 2019-11-13 by rmackay9

jasin755, 2019-11-13:
Please reopen. I found problem. Sbus out is comoletely off until safety switch on. It's can't be bypassed. I think it is issue of documentation.

rmackay9, 2019-11-13:
@jasin755,

Thanks for following up and helping us get to the bottom of this.

Just for reference, this is the appropriate wiki page I think: http://ardupilot.org/copter/docs/common-sbus-out.html

Does changing BRD_SAFETY_MASK doesn't help?  I would expect that the servos's wouldn't move if the BRD_SAFETYENABLE = 1 and appropriate bit in the BRD_SAFETY_MASK wasn't set.  For example if the SBUS servo was on (output) channel 5 then I would expect the "CH5" bit (i.e. 16) would need to be set.