ArduPilot/ardupilot #12832: Unintentional parachute release, due to Sbus glitch:

AngelAlbeniz, 2019-11-14:
I have suffered two unintentional parachute release, the cause of these deploys is a high pulse (greater than 1800) on channel 7, which is assigned the function of "parachute release".
These pulses have a very short duration. In the dataflash logs, which sample RCIN values at 10Hz, these errors are represented by a single sample I'm guessing they can be "glitches" in a single package of the Sbus frame. 

Months ago, I experienced problems with FailSafes and Sbus frames. If the receiver FailSafe was set to “no pulse” it could happen that some channels “returned” from the failsafe before others causing low pulses of short duration that could change the flight mode, disarm the equipment etc. Same problems as #9389 #9400 #10139 which were fixed with #10216. At that moment I fixed it by configuring the failsafe of the receivers in hold mode.

The problem I'm experiencing now is somewhat similar. I have periodic glitches in which one of the 8 channels changes to a low value during a very short period, but this low value is not necessarily less than 900. But also, although it is a stranger event, I have had glitches in which one of the 8 channels changes to a high value, not necessarily higher than 2200. If this happens on channel 7 with a pulse greater than 1800, the parachute fires immediately.

These glitches are not very frequent, I may have many flights in which none are observed. This makes it very difficult to do a bench tests to determine the exact cause of the problem. it may be wireless corrupted packets, a problem in de RX generating the Sbus frame or a bad handling of the Sbus frame and failsafes by the autopilot.

In addition, I have tried different combinations of Frsky radios and receivers (X9D plus, QX7) (XM+, X8R) and I experience the same problem with everyone. 

On the other hand, most of the radio protocols used in the sector are not reliable enough and do not include functionalities such as ECC or EDC, so I think that critical flight and safety functions such as "Parachute Release", "Disarm" or "Emergency Stop Motors" should have some extra security layer for their activation. A combination of different channels, a minimum pulse duration (Ex: 1s) for activation or both. 

Attached some photos of the RCIN graphics of the dataflash logs.

Two glitches that fired the parachute:
[image]
[image]

Other observed glitches:
[image]
[image]
[image]

### Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X

**Hardware type**
Pixhawk and Cube

**Logs**
Please provide a link to any relevant logs that show the issue

peterbarker, 2019-11-14:
On Thu, 14 Nov 2019, AngelAlbeniz wrote:

> On the other hand, most of the radio protocols used in the sector are not reliable enough and do not include functionalities such as ECC or EDC, so I
> think that critical flight and safety functions [...]

We should require three readings to trigger a parachute release, at least 
on master.

> Version
> What version was the issue encountered with

You didn't fill this bit in....

Pedals2Paddles, 2019-11-15:
Is 3 loops long enough?  For functions that will crash the vehicle, it seems like at least a half second to a second would be better.

peterbarker, 2019-11-16:
On Thu, 14 Nov 2019, Matt wrote:

> Is 3 loops long enough? For functions that will crash the vehicle, it seems like at least a half second to a second would be better.

That half-second-to-second might make the difference when actively 
manipulating the vehicle way - i.e. when you deliberately want the vehicle 
to stop spinning its props, the vehicle waiting for another second might 
cause significant extra danger.

Q: was this *really* RC "glitching", or was the RC signal marginal and the 
receiver dropping to its bind-time channel values?

AngelAlbeniz, 2019-11-18:
> That half-second-to-second might make the difference when actively
manipulating the vehicle way - i.e. when you deliberately want the vehicle
to stop spinning its props, the vehicle waiting for another second might
cause significant extra danger.

You are right, the system should have a fast response in this type of critical actions, but it is also very important that it is reliable enough and avoid false positives. At this point I'm questioning the usefulness of using a parachute system which can be shot intentionally and has no security layer for it.

The system requiring three readings to trigger a parachute release, will be nice and better and better than not requiring anything, of course.

>Q: was this *really* RC "glitching", or was the RC signal marginal and the
receiver dropping to its bind-time channel values?

My guess is that there are “glitches”, first because they occur when the signal is good as when it is bad. And second because both the number of channels it affects, and the values taken by each channel is random. If the receiver drops to its bind-time channel values, these values will be always the same, True?

Hasi123, 2020-01-08:
Maybe something like this could be implemented in Copter: https://github.com/ArduPilot/ardupilot/commit/c73a4fd2274e6f3a394eb3ec4fc9109e5ad99584

IamPete1, 2021-08-15:
https://github.com/ArduPilot/ardupilot/pull/11623 should have made the de-bouncing more robust. Please re-open if you encounter any issues on 4.1.

closed 2021-08-15 by IamPete1