ArduPilot/ardupilot #12838: AP_Compass: diagonals and off-diagonals are not saved reliably

rmackay9, 2019-11-15:
It seems that the onboard compass calibration is not reliably saving the COMPASS_ODI_X/Y/Z and COMPASS_DIA_X/Y/Z parameters.

Some information from the investigation so far:

- the compass calibrator itself is failing to calculate the diagonals (i.e. it's not a parameter saving issue)
- if a lower index compass fails to save diagonals, compasses with higher indices will also fail.  I.e. if compass0 fails to calculate a diagonals, compass1 and compass2 will also fail
- the issue can be reproduced in SITL, Copter-3.6 and Copter-4.0
   - param set SIM_MAG_ODI_X 0.05
   - param set SIM_MAG_ODI_Y 0.04
   - param set SIM_MAG_ODI_Y 0.03
   - param set SIM_MAG_DIA_X 1.05
   - param set SIM_MAG_DIA_Y 1.04
   - param set SIM_MAG_DIA_Y 1.02
   - param set SERVO5_FUNCTION = 1 (passthrough)
   - rc 5 1250
   - magcal start

peterbarker, 2019-11-15:
SITL has to be started with `--calibration`

rmackay9, 2019-11-19:
By the way, I tested PR https://github.com/ArduPilot/ardupilot/pull/12177 but it doesn't seem helps with this specific issue at least (perhaps it helps in other ways)

rmackay9, 2019-11-21:
After much investigation we've concluded that the compass diagonals and offdiagonals occasionally not saving is not actually a bug.  Instead we think it's proper behaviour and this is happening simply because the "sphere" is extremely round.  I.e. the diagonals don't help with the calibration

closed 2019-11-21 by rmackay9

lukezhqin, 2019-11-21:
@rmackay9 ,I went through the L-M algorithm but found what you mentioned above could not be a problem. Because we could get a bad convergence if we kept updating the diag or off-diag as you explained. Is there any thing else that I would missed or misunderstood then?

lukezhqin, 2019-11-21:
> @rmackay9 ,I went through the L-M algorithm but found what you mentioned above could not be a problem. Because we could get a bad convergence if we kept updating the diag or off-diag as you explained. Is there any thing else that I would missed or misunderstood then?

Sorry about the query here, That I dind't notice that you have got a conclution just right now.

rmackay9, 2019-11-21:
@lukezhqin, No troubles, thanks very much for responding.  I'm reasonably convinced that the compass calibration is working as expected... our SITL simulator actually allows simulating the compass calibration and even modifying the vituals offsets and scaling (see SIM_MAG_OFS_xxx, SIM_MAG_DIA_xxx and SIM_MAG_ODI_xxx) and then you can actually do a compass cal in the simulator by doing this ... although when I try to do this the vehicle isn't moving for some reason:

- param set SERVO5_FUNCTION = 1 (passthrough)
- rc 5 1250
- magcal start

You've probably seen it but a modified version of your fix is in this PR: https://github.com/ArduPilot/ardupilot/pull/12864 (the last commit is attributed to you)

Pedals2Paddles, 2019-11-21:
But why would this happen only sometimes? I can calibrate and get no DIA and ODI.  Then calibrate the same vehicle in the same spot all over again and get values for both.

rmackay9, 2019-11-21:
@Pedals2Paddles, I think the diagonals and off-diagonals the algorithm is coming up with are  probably very close to 1.0 and 0.0.  They are probably so close that the impact of having them or not having them is very small.  The way we can prove it is if I provide you with a special version that stores the sensor data and then we can visualise the points using https://plot.ly/ and/or we can apply the offsets the calibration comes up with and then compare the lengths to all the individual points.  When I did this yesterday with a compass that was behaving like yours we found the sensor values were appearing in a very nice sphere.

Pedals2Paddles, 2019-11-23:
IIRC they were very small values so you're probably right.