ArduPilot/ardupilot #12841: Mateksys F765-Wing Servos 5/6 inop

c17loadktcm, 2019-11-15:
### Bug report

**Issue details**

I contacted MATEKSYS about Servo's 5/6 not functioning. The response was that it was an Ardupilot issue, https://github.com/ArduPilot/ardupilot/pull/12685 fixed in the latest firmware later than Oct.30.  not the beta of Oct.28

I flashed the latest version for MATEK765-WING (http://firmware.ardupilot.org/Plane/latest/MatekF765-Wing/) last night. I flashed Arduplane.hex, had to flash via betaflight, kept getting json errors in mission planner.

Please let me know if this is in the works or what version i need to flash to the board.

Servo outputs 1-4 and 7-10 work. I get both motors and ailerons on S1-4 and i can get the vtail left and right to work on S7/8 nothing on S5/6

**Version**
4.1.0-FIRMWARE_VERSION_TYPE_DEV

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
MakeFlyEasy Believer V-Tail

**Hardware type**
Mateksys F765-WING

**Logs**
no logs available

c17loadktcm, 2019-11-16:
I've been working with Matek Staff for the last few hours. They had me Flash the latest arduplane_with_BL version 4.1.0 dev. I've attached a YAAPU screenshot of the firmware version, CHibios verson and f765. I was able to flash INAV to the board and confirm that indeed, the two ports do actually function,
[image]

thadeausmaximus, 2019-11-16:
I have the same thing. I contacted Matek support and they said it was working in latest but it doesn't work for me.

c17loadktcm, 2019-11-16:
Well I'm glad it isn't just me. I did verify that there's nothing wrong
with the board by installing INAV firmware. All servo outputs work with
INAV so the problem has to be with ArduPlane. I am not a software engineer,
but my deductive reasoning is, its a code issue. I even had Matek send me
the code they said works, still nothing. I'm just going to map my v-tail
left and v-tail right to S7 and S8 until this gets fixed. Everything else
on the board works with ArduPlane just fine. GPS, Telemetry Radio, Motors,
Frsky Smart port, OLED, FPV cameraDigital Airspeed, Matek buzzer, parachute
release, and camera trigger.

Its really nice compared to the wiring nightmare of a regular pixhawk.

Hopefully we can have this resolved soon.

Brian

c17loadktcm, 2019-11-20:
I can confirm that Arduplane 4.0.1beta1 has restored Servo 5/6 (S5/S6 on the board) function.  This beta was released on 18 November 2019. I'll let the team close this issue when it is implemented in the 4.0.1 stable release. @tridge @rmackay9

rmackay9, 2019-11-20:
Great, I think I'll close this now if that's OK.  That's normally how we do it although there is certainly a good argument for leaving it open until it's released..

closed 2019-11-20 by rmackay9

Rolf-G, 2019-12-22:
Sorry, but i can't confirm Servo5/6 function are restored in V4.0.2beta2. 
MatekF765-Win 00260038 30385102 3935333
ChibiOS: 31fc3131
ArduPlane V4.0.2beta2 (46b36493)

https://discuss.ardupilot.org/t/plane-4-0-release/47188/192

Best Regards Rolf