ArduPilot/ardupilot #12842: Wifi UAS remote identification from dji to others

roque-canales, 2019-11-16:
https://www.dropbox.com/s/v4lkyr2kdp8ukvx/DJI%20Remote%20Identification%20Whitepaper%203-22-17.pdf?dl=0

DJI propose to governments this “open” uas remote identification protocol that will become soonly mandatory for use uas.

Does anyone here have more information to this protocol? Its based on 2.4ghz so maybe connecting little esp to mavlink we can dev low cost solution that use this protocol?

peterbarker, 2020-01-07:
I think events may have overtaken this with the new FAA proposal to use "da internet" to solve the problem.

CraigElder, 2020-01-14:
@apache405

rmackay9, 2020-01-14:
@magicrub - we also thought you might have an opinion on this one.

closed 2020-01-14 by rmackay9

peterbarker, 2020-01-14:
@magicrub and Jeff W.

This is a support request; please open something on discuss.ardupilot.org if you'd like to push further with this.

apache405, 2020-01-14:
@magicrub and I were heavily involved in the development of Open Drone ID, which seems to be what the FAA is basing their requirements on.  DJI was also involved with the development of OpenDroneID (specifically on the WiFi side of things).  Several of us are working on hardware and firmware to address the remote ID requirements.

roque-canales, 2020-01-15:
Hello @apache405, 

Many thanks for this news.

When its planned to have the solution working with hardware?

Here in france all uavs must be compliant before julliet....(for the wifi side)