ArduPilot/ardupilot #12853: Latest build fails (AP_KDECAN errors)

msadowski, 2019-11-18:
### Bug report

I just pulled the latest changes on master (89c43fa052865ce576c9e4d24d54eab9654de0e2) and tried building them for fmuv3 (building for copter). The error shown was:

```
[201/639] Compiling libraries/AP_Motors/AP_MotorsTailsitter.cpp
../../libraries/AP_KDECAN/AP_KDECAN.cpp: In member function 'virtual void AP_KDECAN::init(uint8_t, bool)':
../../libraries/AP_KDECAN/AP_KDECAN.cpp:75:6: error: '%u' directive output may be truncated writing between 1 and 3 bytes into a region of size 2 [-Werror=format-truncation=]
 void AP_KDECAN::init(uint8_t driver_index, bool enable_filters)
      ^~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors

Waf: Leaving directory `/home/mat/Consulting/Terabee/ardupilot/build/fmuv3'
Build failed
 -> task in 'objs/AP_KDECAN' failed (exit status 1): 
	{task 139848517697240: cxx AP_KDECAN.cpp -> AP_KDECAN.cpp.0.o}
 (run with -v to display more information)
```

This might be related to https://github.com/ArduPilot/ardupilot/pull/12789

**Version**
Commit: 89c43fa052865ce576c9e4d24d54eab9654de0e2

**Platform**
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**
N/A

rmackay9, 2019-11-19:
@msadowski, thanks for the report.  Which board target were you building for?  I'm not seeing this failure with master but that doesn't necessarily mean that there isn't a problem.

peterbarker, 2019-11-19:
On Mon, 18 Nov 2019, Mateusz Sadowski wrote:

> I just pulled the latest changes on master (89c43fa) and tried building them for fmuv3 (building for copter). The error shown was:

I'm not able to reproduce locally.

Which compiler?

msadowski, 2019-11-19:
Funnily enough I can't reproduce it either. I think the issue was with my build process. Here is roughly how I think the process might have went:

- pulled the repo, tried to build for master (fmuv3) (didn't work due to errors)
- checked out f0d5929408599b453 (latest release)
- Try building again (I've also seen errors)
- Realized I did the install and am probably missing some tools - installed them using the script recommended in the docs
- f0d5929 started building fine
- There was still an error on the master, however after I removed the build directory and did ./waf configure again it seems to be building again.

I'll close this for now, If I run into any issues moving on I'll update this Issue.

closed 2019-11-19 by msadowski