ArduPilot/ardupilot #12862: Watchdog reset due to interrupt storm\\\\\\\\\\\\

Hwurzburg, 2019-11-20:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

tridge, 2019-11-23:
I have reproduced this by connecting telem1 TX in parallel with the SDA line of i2c2 on a MatekF405-Wing. I am working on a fix

tridge, 2019-11-23:
ChibiOS bug report here:
http://www.chibios.com/forum/viewtopic.php?f=35&t=5298

tridge, 2019-11-26:
fixed by #12896

closed 2019-11-26 by tridge