ArduPilot/ardupilot #12876: Subgroupinfo of subgroupinfo

xxxmonsterxxx, 2019-11-24:
**Bug report**
So, i tried to add AC_PID class object into AP_MotorsHeli_RSC library and got problem with #include<AC_PID/AC_PID.h> preprocessor procedure... In compilation got error about AP_Logger::PID_Info is does not name a type. I think it's problem of hierarchy of includes. This problem could be solved using incomplete type of AC_PID and pointer instead object.. but i hope somebody will give me right way to solve this problem.

Thank you.

**Version**
master
**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

xxxmonsterxxx, 2019-11-24:
And next question here, cause i would not to create extra issue. After i did add subgroupinfo into parameters of library which already is subgroup of other library, apm planner wan't load all (about 700, but i know all params more than 1k) parameters from pixhawk. So now question, it's bug, or iface of subgroupinfo cannot make this thing?

magicrub, 2019-11-25:
Please post a link to your branch.

xxxmonsterxxx, 2019-11-25:
> Please post a link to your branch.

Hi, sorry, but i haven't possibility for pushing my local branch, bad inet connection. But, i could to write some realization. My local branch was forked from master. Which my code is interests for you?

bnsgeyer, 2019-11-25:
@xxxmonsterxxx hi, this is not really a bug report or an issue with the code. It sounds like you’re trying to do something new and don’t know how to code to change. If you don’t mind me asking, please tell me what you are  trying to incorporate into the code.

xxxmonsterxxx, 2019-11-25:
> @xxxmonsterxxx hi, this is not really a bug report or an issue with the code. It sounds like you’re trying to do something new and don’t know how to code to change. If you don’t mind me asking, please tell me what you are trying to incorporate into the code.

hi =), simple plan. I tried to add pid object into rsc controller. And then params of pid add into rsc params as subgroupinfo.
I tried so...
AP_MotorsHeli_RSC.h
#include <AC_PID/AC_PID.h>
.....
AC_PID _pid_object;
_____________________________
AP_MotorsHeli_RSC.cpp
...
Var_info:
...
AP_SUBGROUPINFO(_pid_object, "PID_", 18, AP_MotorsHeli_RSC, AC_PID),

thank you!

bnsgeyer, 2019-11-25:
@xxxmonsterxxx i understand. I’m just wondering why you want to do that?  What is your ultimate goal?  The RSC contains a governor which doesn’t conform to the standard PID structure which is why we don’t use a PID object.  Just looking for some idea of your end goal.

xxxmonsterxxx, 2019-11-25:
> @xxxmonsterxxx i understand. I’m just wondering why you want to do that? What is your ultimate goal? The RSC contains a governor which doesn’t conform to the standard PID structure which is why we don’t use a PID object. Just looking for some idea of your end goal.

The ultimate goal is the governor as the pid controller. But it is not important. This is just a small idea to compare the quality of the pid and the already implemented governor.
I'm more interested not in goal but in process of implementation of this.

xxxmonsterxxx, 2019-11-25:
> @xxxmonsterxxx i understand. I’m just wondering why you want to do that? What is your ultimate goal? The RSC contains a governor which doesn’t conform to the standard PID structure which is why we don’t use a PID object. Just looking for some idea of your end goal.

I would be glad, if you could to say me, why my realization is wrong? Because i get compilation error.

bnsgeyer, 2019-11-25:
@xxxmonsterxxx you will need to give me a link to your branch in order for me to better help fix your problem.  It is not that easy to fix though messages.  You can look at examples in other parts of the code where they have done similar things. 
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_AttitudeControl/AC_AttitudeControl_Heli.cpp

xxxmonsterxxx, 2019-11-26:
> @xxxmonsterxxx you will need to give me a link to your branch in order for me to better help fix your problem. It is not that easy to fix though messages. You can look at examples in other parts of [...]

Thank you, but i did same...
Okay, ill try to push my branch and give to you the link.

xxxmonsterxxx, 2019-11-26:
> @xxxmonsterxxx you will need to give me a link to your branch in order for me to better help fix your problem. It is not that easy to fix though messages. You can look at examples in other parts of [...]

https://github.com/xxxmonsterxxx/ardupilot/tree/pid_gov
see please this implementation.

bnsgeyer, 2019-11-26:
One thing you left out was a statement in the constructor that contains the defaults.  I've shown an example from the attitude controller for heli.
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_AttitudeControl/AC_AttitudeControl_Heli.h#L32-L42

I think we tried doing this about a year ago and remember getting a similar error that you are getting.  I'm not sure what is causing it.  I'm wondering if it has something to do with the fact that RSC is also an object and not a library like MotorsHeli.  try adding a PID object to MotorsHeli and see if you get the same error.  I am not that good with C++ programming.

xxxmonsterxxx, 2019-11-26:
> One thing you left out was a statement in the constructor that contains the defaults. I've shown an example from the attitude controller for heli. [...]

Thbank you so much for constructive answer.
About subgroupinfo bug... Yeah, i suggested same. But ds_PID in https://github.com/xxxmonsterxxx/ardupilot/blob/master/libraries/AP_Landing/AP_Landing_Deepstall.cpp and onetime deepstall in https://github.com/xxxmonsterxxx/ardupilot/blob/master/libraries/AP_Landing/AP_Landing.cpp
I don't know if this work or not... but this code in master...

About compilation error with #include AC_PID... i'll try. About default constructor - doesn't matter, it's just was example.

xxxmonsterxxx, 2019-11-27:
PID doesn't include into MotorsHeli also. So, i think it's BUG of "include hierarchy" and should be fixed.

bnsgeyer, 2019-11-27:
There may be something wrong with the code that it doesn’t allow you to incorporate the PID library. However it currently doesn’t break any features that currently exist in the code and therefore wouldn’t get any attention to be fixed. Plus in my opinion there’s no need to access any of the PID library‘s. The only feature that would use this would be the governor and the way it’s currently implemented it does not need a PID library.

xxxmonsterxxx, 2019-11-27:
> There may be something wrong with the code that it doesn’t allow you to incorporate the PID library. However it currently doesn’t break any features that currently exist in the code and therefore [...]

Okay, thank you for contribution in this issue.

closed 2019-11-27 by xxxmonsterxxx