ArduPilot/ardupilot #12910: Analog airspeed sensor calibration

dibelardino, 2019-11-27:
### Bug report

@tridge

**Issue details**
Good evening.
I have an issue calibrating an analog airspeed sensor.
Since the airplane is as fast as 225 km/h I noticed that, at higher speeds, there was a difference between the AS and the GS.
The auto calibration was not able to fix the difference.

_Please describe the problem_

Looking at the code and the datasheet of the analog sensor MPXV7002DP I found that the code contains a wrong VOLTS_TO_PASCAL constant definition.
Looking at the datasheet (see the attached image) the voltage to pascal factor should be 1000 instead of 819 defined in the AP_Airspeed_analog.
Since the wrong pressure conversion factor is under square root, the error is not linear and can’t be corrected with the airspeed ratio.
At the normal model airspeed the error is not evident, but it is flying at 200 km/h.
Since the pressure formula is 
pressure = _source-> () * VOLTS_TO_PASCAL / get_psi_range();
I modified the value of ARSPD_PSI_RANGE,1 to 0.819 correcting the wrong VOLTS_TO_PASCAL value.
It would be advisable to modify the #define in the code, but it would mean recalibrating all the airspeed sensor after the new release installation.
By the way, since the range of the MPXV7002DP is 2 kPa that should correspond to 200 km/h, I switched to the MPXV5004 sensor that has a range of 4 kPa.

**Version**
_What version was the issue encountered with_
last

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Ultralight aircraft

**Hardware type**
 

Pixhawk

**Logs**
_Please provide a link to any relevant logs that show the issue_

[image]

robustini, 2019-11-29:
I think @tridge can help you with this.

dibelardino, 2019-11-30:
Little Update.
Today I was able to fly again.
With the following settings:
ARSPD_OFFSET,1215
ARSPD_PSI_RANGE,0.819
ARSPD_RATIO,1.5
Good airspeed reading, stable and correct from 50 km/h up to 200 Km/h

IamPete1, 2021-09-06:
It sounds like setting `ARSPD_PSI_RANGE` is a valid solution?

Its a shame we don't have a direct param, I don't think worth adding a new param just for that.

closed 2021-09-22 by IamPete1