ArduPilot/ardupilot #12913: Copter: pre-arm check "Terrain data missing" when attempting to arm without GPS lock

rmackay9, 2019-11-27:


rmackay9, 2019-12-14:
It's possible that this is caused by the terrain failsafe and probably only appears if the vehicle is in Auto, Guided or RTL mode while disarmed.

peterbarker, 2020-01-07:
You're suggesting this message should be squelched, I assume :-)

rmackay9, 2020-01-07:
Yes, this warning is misleading I think.  Saying that, I haven't seen it again myself nor has any user complained.

IamPete1, 2021-08-15:
The cause is terrain failsafe, not related to arming directly.

https://github.com/ArduPilot/ardupilot/blob/25ee57080fb6f86487d67ad2c55f3b79e046ad67/ArduCopter/events.cpp#L270

I guess the correct fix would be to refuse to arm if were just going to instantly hit terrain failsafe.

DanishVasdev, 2022-01-26:
Hi, I would like to work on this issue. Could you guide me on how would I reproduce this on SITL?...I am new to ardupilot so I am not sure exactly where would I find this
I tried using the "testing GPS failure" given in -
https://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html
But here arming with this I couldn't see any such message on the console...only "pre arm fail"  and "AP: Pre Arm: Bad Fix"

Jai-GAY, 2024-10-17:
I got this message when testing a LUA Script. A day earlier, it was okay and I could repeatedly execute the script without any hardware or software reboot.

[CubeOrange, 4.4.4.](https://drive.google.com/file/d/1X6W0VLPWYVN5WgjdgIVD_Z5zWXTN7b5W/view?usp=drive_link)
```
GPS 1: specified as   UAVCAN1-125
Arming -> Taking off RC11 : 2
Taking off
Failsafe: Terrain data missing
Wait for arming : 2
```

Jai-GAY, 2024-10-18:
get it again, [same 4.4.4, same script](https://drive.google.com/file/d/1kdMMJTnj2PhJRou0gc8TI0n_AndqmQ4L/view?usp=drive_link), on different day.