ArduPilot/ardupilot #12917: Copter/Plane: handle QGC command uploads alt MAV_FRAME_GLOBAL_TERRAIN_ALT_INT

rmackay9, 2019-11-28:
ArduPilot Copter and Plane only accept mission commands with terrain altitudes specified as MAV_FRAME_GLOBAL_TERRAIN_ALT ("10") but QGC uploads these using MAV_FRAME_GLOBAL_TERRAIN_ALT_INT ("11").  We should accept either.

At the moment, if the user attempts to upload using QGC, the altitude will silently be changed to an absolute altitude which is bad behaviour in itself actually.

The difficulty may come when we have to reply to the ground stations consistently with what they have uploaded (i.e. reply with TERRAIN_ALT_INT if that's what they uploaded)

WickedShell, 2019-11-30:
I'd have to double check but historically we always respond in whatever we think it should be regardless of the GCS request, which isn't the worst thing. Frankly the `_INT`'s were a bad idea.

Anyways I'd agree that the 2 obvious things to do at the moment (that should be fairly quick):
1. Translate TERRAIN_ALT_INT to TERRAIN_ALT internally  on upload
2. NACK the uploaded item if it's sent with a frame we don't understand. (`MAV_MISSION_UNSUPPORTED_FRAME` is for exactly this purpose)

peterbarker, 2019-12-03:
@DonLakeFlyer I think this might be an issue in QGC.

It definitely seems to be uploading with an incorrect frame.

Take this mission:
```
{
    "fileType": "Plan",
    "geoFence": {
        "circles": [
        ],
        "polygons": [
        ],
        "version": 2
    },
    "groundStation": "QGroundControl",
    "mission": {
        "cruiseSpeed": 15,
        "firmwareType": 3,
        "hoverSpeed": 5,
        "items": [
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 50,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 1,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    0,
                    -35.36288833618164,
                    149.164794921875,
                    50
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": 2386,
                "Altitude": 1802,
                "AltitudeMode": 3,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 2,
                "frame": 0,
                "params": [
                    0,
                    0,
                    0,
                    0,
                    -35.36267852783203,
                    149.1656951904297,
                    2386
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 600,
                "AltitudeMode": 2,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 3,
                "frame": 0,
                "params": [
                    0,
                    0,
                    0,
                    0,
                    -35.363525390625,
                    149.16580200195312,
                    600
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 50,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 4,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    0,
                    -35.36398696899414,
                    149.16468811035156,
                    50
                ],
                "type": "SimpleItem"
            }
        ],
        "plannedHomePosition": [
            -35.36326217651367,
            149.1652374267578,
            585
        ],
        "vehicleType": 2,
        "version": 2
    },
    "rallyPoints": {
        "points": [
        ],
        "version": 2
    },
    "version": 1
}
```

When I added this debug:
```
@@ -551,6 +553,7 @@ void PlanManager::_handleMissionRequest(const mavlink_message_t& message, bool m
                                                item->param7(),
                                                _planType);
     } else {
+        ::fprintf(stderr, "itemx %u has frame %u\n", (unsigned)missionRequest.seq, (unsigned)item->frame());
         mavlink_msg_mission_item_pack_chan(qgcApp()->toolbox()->mavlinkProtocol()->getSystemId(),
                                            qgcApp()->toolbox()->mavlinkProtocol()->getComponentId(),
                                            _dedicatedLink->mavlinkChannel(),
```
in `PlanManager.cpp` I get this output:
```
itemx 0 has frame 0
itemx 1 has frame 3
itemx 2 has frame 0
itemx 3 has frame 0
itemx 4 has frame 3
```

One of those items should have a frame of 10 - `TERRAIN_ALT`.

The frame is zero in the saved file, but reloading the file does seem to work - I guess it's going off the other fields in the JSON.

@DonLakeFlyer should I raise an issue against QGC?

DonLakeFlyer, 2019-12-10:
QGC doesn't support MAV_FRAME_GLOBAL_TERRAIN_ALT_INT. "Above Terrain" for altitudes sets an absolute altitude based on the terrain height plus the request all above terrain.

rmackay9, 2019-12-10:
I'll close this issue then 'cuz we've got the issue now on the QGC issues list.  Thanks @DonLakeFlyer!

closed 2019-12-10 by rmackay9