ArduPilot/ardupilot #12920: Kakute F7 v1.2 AC4.0.0rc2 enable compass box greyed out

FilippoFranchini, 2019-11-28:
### Bug report

I have a drotek M8T + compass LIS3MDL. When I install AC3.6.11 the compass works fine while if I install AC4.0.0 I have BAD AHRS and compass not healthy messages. The box enable compass is greyed out and if I try to click it the following message appears: "This feature is not supported with your firmware".

_Please describe the problem_

**Version**
AC4.0.0 rc2

**Platform**
Copter

**Airframe type**
Quad

**Hardware type**
Kakute F7 1.2, M8T + LIS3MDL GPS-compass module, R9slim 

**Logs**
NA

andyp1per, 2019-11-28:
Duplicate of https://github.com/ArduPilot/ardupilot/issues/12512

closed 2019-11-28 by andyp1per

rmackay9, 2019-11-29:
It's actually not the same issue but in any case, I've raised a request on MP to remove the "Enable compasses" checkbox on the compass setup screen. https://github.com/ArduPilot/MissionPlanner/issues/2264