ArduPilot/ardupilot #12923: Terrain data missing for finland

rmackay9, 2019-11-28:
During investigation of [this user report](https://github.com/ArduPilot/ardupilot/issues/12888), I found that the terrain data seems to be missing for Finland () which I think means that terrain following missions will not work.

@meee1 tells me that there should be a N62E025.hgt file in [this directory](http://firmware.ardupilot.org/SRTM/Eurasia/) but there are not files higher than N60.

thadeausmaximus, 2019-11-30:
The terrain data comes from the SRTM, Shuttle Radar Topography Mission, and only covers from 60 N about 55 S. So you are north of the terrain database used. As the old cartographers said "There be dragons." Unless you have a line on good terrain data for that far north you may be out of luck.

rmackay9, 2019-12-02:
@thadeausmaximus, thanks for this.

I've just done a quick test in SITL (both the regular one using mavproxy and the one built into MP) and I find that the terrain failsafe did not trigger when the user enters a waypoint with an altitude frame specified as "Terrain" (aka AGL).

VDLJu, 2019-12-02:
There is reliable terrain data available from link below as WMS and WCS services. The map layer is "paituli:mml_korkeusmalli_10m_2019".

The grid size is 10m x 10m and the vertical accuracy is about 1,4 m. Height is given in N2000 system. It is the most accurate DEM available for whole Finland
Coordinate Reference System: ETRS89 / ETRS-TM35FIN (EPSG:3067)

At current state the difference/error between N2000 system and WGS84 (gps default) heights are under 1m. IMO this could be used as is, without any conversion.

API info to database
https://avaa.tdata.fi/web/paituli/rajapinta

Description for the DEM
http://www.nic.funet.fi/index/geodata/mml/dem10m/2019/MML_10mDEM_description.pdf

rmackay9, 2019-12-03:
Mission Planner (beta) has been updated so that it will not send invalid altitudes when it has no SRTM data.  I think this will resolve the terrain-failsafe issue (i.e. a terrain failsafe will be triggered)

tridge, 2024-04-25:
yes, issue is now resolved

closed 2024-04-25 by tridge