ArduPilot/ardupilot #12929: Monitor uavcan GPS status in real time

cuhome, 2019-11-29:
After I disconnected the UAVCAN GPS from the flight controller, the gpsstatus still shows as 1; during the flight, UAVCAN GPS may not work / cable is abnormal / GPS positioning is lost, and even worse, GPS2 no matter what you use GPS, it can not monitor the GPS connection status in real time. More and more drone jammers on the market may require more GPS information to help manufacturers analyze the source of the failure.
[image]

tridge, 2019-11-29:
what version of ArduCopter? I fixed this in a recent change

cuhome, 2019-11-29:
AC4.00 DEV

cuhome, 2019-11-29:
[image]

tridge, 2019-11-29:
That version is from October 10th. I put in the fix on October 14th

cuhome, 2019-11-29:
Thanks, I have tested the latest firmware and it does.

cuhome, 2019-11-29:
Test completed, thank you.

closed 2019-11-29 by cuhome