ArduPilot/ardupilot #12948: Plane: Flight timer on OSD (MAX7456) displays random digits after some time

giacomo892, 2019-12-02:
After 67:07 minutes of flight the flight timer duration on the OSD started to show random values , using lot of space and covering other symbols as you can see from the current screenshot:

[image]

Video showing the issue:

[2019-12-02 10.38.19.mp4.zip](https://github.com/ArduPilot/ardupilot/files/3910507/2019-12-02.10.38.19.mp4.zip)

**Version**
Running Master at fba5732292 compiled on OSX using the suggested compiler from brew.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Matek F405 STD (ChibiOS) 

**Logs**
Have been sent directly to @tridge for privacy reasons.

tridge, 2019-12-02:
should be fixed by #12949

giacomo892, 2019-12-07:
@tridge not fixed. It happened again on my setup running latest master. This time it happened at 68.27 minutes of flying so around the same of the previous flight. You should get / already got some useful data to further debug the issue. Thanks

peterbarker, 2019-12-07:
@giacomo892 are you able to test https://github.com/ArduPilot/ardupilot/pull/13001 ?

That roughly-an-hour time is quite a tell - 16-bit milliseconds wraps at that point.

giacomo892, 2019-12-07:
@peterbarker 
It's very unlikely for that to be the issue, since I have multiple reports from a friend who has flown much more than me and had no issues in the timer.

This time I've noticed that once this happened the messages showed "mission 3 WP" briefly but I was in FBWA!

Then I landed.. tried to arm again and  I was getting prearm internal errors 0x8000

peterbarker, 2019-12-07:
@giacomo892 OK, before you landed you should have been being screamed at - internal errors are very bad.

Does tridge have the logs for that internal-error flight?

tridge, 2019-12-08:
The log is in AP_Stat
2019-11-30 22:23:13.07: PARM {TimeUS : 4158508040, Name : STAT_FLTTIME, Value : 5763.0}
2019-11-30 22:23:43.07: PARM {TimeUS : 4188510464, Name : STAT_FLTTIME, Value : 5793.0}
2019-11-30 22:24:14.07: PARM {TimeUS : 4219507674, Name : STAT_FLTTIME, Value : 5824.0}
2019-11-30 22:24:45.07: PARM {TimeUS : 4250508090, Name : STAT_FLTTIME, Value : 5855.0}
2019-11-30 22:25:16.07: PARM {TimeUS : 4281507393, Name : STAT_FLTTIME, Value : 5886.0}
2019-11-30 22:25:46.15: PARM {TimeUS : 4311586762, Name : STAT_FLTTIME, Value : 1979985792.0}
2019-11-30 22:26:17.15: PARM {TimeUS : 4342586833, Name : STAT_FLTTIME, Value : -17173922.0}
2019-11-30 22:26:47.16: PARM {TimeUS : 4372598201, Name : STAT_FLTTIME, Value : -841807616.0}

It is clear that STAT_FLTTIME goes crazy within 30s of 5886.

giacomo892, 2019-12-08:
@tridge do you have an idea why message box on the OSD displayed again "mission 3 WP"  ? 
It had displayed it after the plane was heading to WP3 after AUTO takeoff just before I switched to CRUISE and continued my flight normally till the point the timer went crazy and displayed that message briefly again.

tridge, 2019-12-08:
I think it is because MatekF405-STD user a 16 bit timer

tridge, 2019-12-08:
it is clearly wrap of 32 bit micros boot time:
2019-11-30 22:25:16.07: PARM {TimeUS : 4281507393, Name : STAT_FLTTIME, Value : 5886.0}
2019-11-30 22:25:46.15: PARM {TimeUS : 4311586762, Name : STAT_FLTTIME, Value : 1979985792.0}

4281507393 0xFF329E41 037714517101
4311586762 0x100FD97CA 040077313712

tridge, 2019-12-08:
I've reproduced the issue, still working on a fix

PR #12949 (2019-12-03): AP_Stats: added semaphore to flttime update
fixes #12948
commit 0dd64150 (2019-12-03): AP_Stats: added semaphore to flttime update

fixes #12948
PR #13002 (2019-12-09): HAL_ChibiOS: fixed handling of 16bit timer wrap
this fixes #12948
This is a critical fix for MatekF765 and MatekF405, which use 16 bit timers
This adds a common system_time_u32_us() so 16 bit and 32 bit systems have identical behaviour
commit 4f429e6c (2019-12-09): HAL_ChibiOS: fixed handling of 16bit timer wrap

this fixes #12948
closed 2019-12-09 by tridge