ArduPilot/ardupilot #12958: Unable to set FS_EKF_THRESH to zero

Vishranthee, 2019-12-03:
Primary , secondary and tertiary EKF values were at 1 soI tried to bring it down so that I can set it at 0.6. But I am unable to do so. Error message: Parameter cannot be set to 0.

What version was the issue encountered with : 3.6.9 build

Quadcopter standard F450 frame

Pixhawk flight controller

rmackay9, 2019-12-03:
I think this is probably a support issue that should go into the forums.  In any case, I just tested Copter-3.6.9 on a CubeBlack (which is close to a pixhawk) and I was able to set the FS_EKF_THRESH to zero without any particular problems.

My guess is that the ground station was refusing to all the parameter to be set to zero but normally user can get around this by using a Full Parameter List or Full Parameter Tree page that doesn't enforce the limits.

I was using the MP ground station during my testing.

I think I will close this issue if that's OK.

closed 2019-12-03 by rmackay9

Vishranthee, 2019-12-03:
I tried to reboot and change it in Full Parameter Tree and it worked. Thanks !!!