ArduPilot/ardupilot #12961: QGC using MISSION_ITEM rather than MISSION_ITEM_INT

peterbarker, 2019-12-03:
### Bug report

**Issue details**

QGC is uploading items using mission_item rather than mission_item_int.

This is bad from an item accuracy POV.

**Version**
master ArduPilot, master QGC

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X

**Hardware type**
SITL

peterbarker, 2019-12-03:
@DonLakeFlyer for visibility

WickedShell, 2019-12-04:
@peterbarker what version did you use for this? The log provided in #12962 uses MISSION_ITEM_INT so I'm starting to get confused how/what versions of QGC people are testing against

peterbarker, 2019-12-04:
On Tue, 3 Dec 2019, WickedShell wrote:

> @peterbarker what version did you use for this? The log provided in #12962 uses MISSION_ITEM_INT so I'm starting to get confused how/what versions of QGC
> people are testing against

I'm using master ArduPilot / master QGC.

I was attaching it to an already-running SITL instance using :14450 - 
Randy was running directly to a vehicle.

My guess is that the AUTOPILOT_VERSION negotiation stuff isn't working 
quite right.  Any time we drop back to MISSION_ITEM is a bug IMO.

peterbarker, 2020-01-07:
@DonLakeFlyer  has fixed this in master QGC - thanks!

closed 2020-01-07 by peterbarker