ArduPilot/ardupilot #12996: BUG with auto_state.checked_for_autoland

liang-tang, 2019-12-06:
### Bug report
Switch AUTO->RTL again an again,  because checked_for_autoland always set to false when change mode

```
bool Mode::enter()
{
    // cancel inverted flight
    plane.auto_state.inverted_flight = false;

    // don't cross-track when starting a mission
    plane.auto_state.next_wp_crosstrack = false;

    // reset landing check
    //plane.auto_state.checked_for_autoland = false;

```

**Version**
4.0
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[*] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2022-03-14:
my be related to https://github.com/ArduPilot/ardupilot/issues/18496 ? In that case you can get stuck switching modes endlessly if you have a `DO_LAND_START` before a `RETURN_TO_LAUNCH` waypoint.

AlSharji0, 2025-07-01:
Hi all,
This is my first contribution to ArduPilot, I've opened PR #30513 to address this issue.

I'd really appreciate any feedback or suggestions for improvement.

Thanks!

AhmetMert01, 2026-10-18:
Hi, I’d like to continue investigating this issue. Following the request for reproducible SITL steps in #30513, I have prepared local autotests that reproduce the landing-sequence restart with RTL_AUTOLAND=1 and 2.
Before proposing a C++ fix, should a landing sequence reaching a mission RTL remain in RTL without restarting landing, or should this be addressed through #18496?
I can share the test patch and logs. Thanks!

anand2532, 2026-11-11:
Hey folks, I tried to recreate this on current master in Plane SITL, and I think it can be done this way.

With RTL_AUTOLAND=0, commanding RTL stays in RTL. With RTL_AUTOLAND=2 and a DO_LAND_START in the mission, I commanded RTL three separate times. Each time the mode went RTL, then straight to AUTO, with Landing sequence start. A mission of takeoff, DO_LAND_START, then RETURN_TO_LAUNCH then printed Landing sequence start over and over, thousands of times in under a minute. The first RTL to AUTO switch looks like the autoland feature. The later ones are the loop.

anand2532, 2026-11-11:
Looking at ModeRTL::navigate(), the successful jump switches to AUTO and returns before checked_for_autoland is set. The flag starts out false, so clearing it in Mode::enter() is not what keeps the loop alive. Removing that clear, as in #35051 , still leaves the flag false on the next RTL.

anand2532, 2026-11-11:
Would this work?

- Set checked_for_autoland = true before both set_mode(AUTO) calls in mode_rtl.cpp, including the return-path branch.
- Stop clearing it in Mode::enter().
- Clear it in AP_Arming_Plane::arm() after a successful arm, so the next flight can still autoland.

The first RTL of a flight would still start the landing sequence. A later RTL would stay in RTL, and a RETURN_TO_LAUNCH mission item could no longer bounce the mode. Happy to put up a patch with a SITL autotest if this is the behavior you want.