ArduPilot/ardupilot #13000: ARMING_REQUIRE ignored

Hwurzburg, 2019-12-07:
Vehicle: copter quadplane... from limited SITL testing, this apparently only happens with Q_TAILSIT_MOTMX nonzero for copter quadplanes...

arming_require = 1 (thr_min before arming)

if no rc, booting to MANUAL no pulse are output to ESCs instead of THR_MIN...
if switched after booting to a VTOL mode pulses begin

if booted to initial_mode = VTOL, pulses begin immediately

this cause some ESCs to lock up until power cycled...they look for valid PWM for a while after boot and will lock if none is provided as a failsafe...

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

tridge, 2019-12-07:
I think we need to change here:
https://github.com/ardupilot/ardupilot/blob/master/ArduPlane/tailsitter.cpp#L101
adding:
        } else {
            motors->output_motor_mask(0, mask, 0);

cohen39, 2020-03-19:
It appears this is no longer an issue and was fixed by commit 90494c9d. I've tested a recent commit (bb0920d5) and code that I believe Hwurzvurg had issues with (ab66b9e9). For a quadcopter quadplane tailsitter, servo channels 5-8 are fed a non-zero PWM in the latest code. In the older commit I was able to replicate the issue and noticed servo channels 5-8 frequency was 0.

Commit ab66b939:
[image]

 Commit bb0920d5:
[image]

Parameters used:
`AHRS_EKF_TYPE    2
ARMING_RUDDER    2
ARMING_REQUIRE   1
ARSPD_FBW_MAX    35
ARSPD_FBW_MIN    13
ARSPD_USE        1
AUTOTUNE_LEVEL   8
COMPASS_OFS2_X   -0.420265
COMPASS_OFS2_Y   -0.726942
COMPASS_OFS2_Z   6.665476
COMPASS_OFS3_X   -0.420265
COMPASS_OFS3_Y   -0.726942
COMPASS_OFS3_Z   6.665476
COMPASS_OFS_X    -0.453570
COMPASS_OFS_Y    -0.585846
COMPASS_OFS_Z    6.815743
EK2_ENABLE       1
EK2_IMU_MASK     3
FBWB_CLIMB_RATE  10
FLTMODE_CH       5
INITIAL_MODE     0
FLTMODE1         0
FLTMODE2         0
FLTMODE3         0
FLTMODE4         0
FLTMODE5         0
FLTMODE6         0
INS_ACC2OFFS_X   0.001000
INS_ACC2OFFS_Y   0.001000
INS_ACC2OFFS_Z   0.001000
INS_ACC2SCAL_X   1.001000
INS_ACC2SCAL_Y   1.001000
INS_ACC2SCAL_Z   1.001000
INS_ACCOFFS_X    0.001000
INS_ACCOFFS_Y    0.001000
INS_ACCOFFS_Z    0.001000
INS_ACCSCAL_X    1.001000
INS_ACCSCAL_Y    1.001000
INS_ACCSCAL_Z    1.001000
INS_GYR_CAL      0
KFF_RDDRMIX      0.500000
LIM_PITCH_MAX    3000
LIM_PITCH_MIN    -3000
LIM_ROLL_CD      6500
NAVL1_PERIOD     14
PTCH2SRV_D       0.309954
PTCH2SRV_I       0.425000
PTCH2SRV_IMAX    4000
PTCH2SRV_P       3.646518
PTCH2SRV_RLL     1
Q_ANGLE_MAX      4500
Q_ASSIST_SPEED   18
Q_ENABLE         1
Q_FRAME_CLASS    1
Q_FRAME_TYPE     0
Q_TAILSIT_MOTMX  7
RALLY_INCL_HOME  0
RALLY_LIMIT_KM   5
RALLY_TOTAL      0
RLL2SRV_D        0.141009
RLL2SRV_I        0.425000
RLL2SRV_IMAX     4000
RLL2SRV_P        1.600000
SCHED_LOOP_RATE  300
THR_MAX          123
THR_MIN          4
TRIM_ARSPD_CM    2500
LOG_BITMASK      65534`

Hwurzburg, 2020-03-19:
thanks! sounds like we can close this one...thanks for testing!

closed 2020-03-19 by Hwurzburg