ArduPilot/ardupilot #13014: ros slam can't arm the copter

kenhuyang, 2019-12-09:
### Bug report

**Issue details**

the link here https://ardupilot.org/dev/docs/ros-connecting.html#ros-connecting
describes how to use rplridar to use slam to fly a copter indoor, i followed the instruction, and the copter in the map moves accordingly.  but when i tried to arm it, the mp says compass not healthy, how to solve this? thanks!

**Version**
3.6.11 version

**Platform**
[  ] All
[  ] AntennaTracker
[ yes ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad x

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
pixhawk
**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2019-12-14:
Could you download a dataflash log and if possible set the LOG_DISARMED = 1 so we can see what it looks like before arming.

IamPete1, 2021-07-26:
No update from @kenhuyang so I'm going to close.

closed 2021-07-26 by IamPete1