ArduPilot/ardupilot #13018: AP_Mission: SITL -- DO_MOUNT_CONTROL handling

RoelHelsen, 2019-12-09:
### Bug report

**Issue details**
The SITL simulation crashes when the flightplan contains a DO_MOUNT_CONTROL (id #205) item. This is due to the type() function in libraries/AP_Mission/AP_Mission.cpp. In the default clause of the switch statement, there is the following code:
`
#if CONFIG_HAL_BOARD == HAL_BOARD_SITL
        AP_HAL::panic("Mission command with ID %u has no string", id);
#endif
`
Since there is no case for the DO_MOUNT_CONTROL, the switch statement defaults and triggers the panic() function, this causes the code to kill the simulation while there is no need for it, as it is a valid mission item. 

Maybe it is worth to consider removing or changing this code, or add cases for all valid mission items.

**Version**
Copter-4.0, the issues was not there in Copter-3.6.10. I think the panic() call was added when moving to Copter-4.0

**Platform**
[ x ] Copter, only in SITL

RoelHelsen, 2019-12-09:
a possibility is to replace the panic() call by the following:
` hal.console->printf("Mission command with ID %u has no string\n", id);`

peterbarker, 2019-12-09:
On Mon, 9 Dec 2019, RoelHelsen wrote:

> Copter-4.0, the issues was not there in Copter-3.6.10. I think the panic() call was added when moving to Copter-4.0

The panic is there to catch exactly this bug.  Working as intended :-)

peterbarker, 2019-12-09:
@RoelHelsen Sorry, I should have been clearer here - I've created a pull request to fix the lack-of-string thing and referenced it in this issue.  I appreciate the issue creation - the panic is there to kind of force the issue, which is, I admit, somewhat heavy-handed!

closed 2019-12-10 by peterbarker

RoelHelsen, 2019-12-10:
@peterbarker So the goal is to find all missing mission items in the switch statement?
Thanks for the quick response and fixing it for everyone!

RoelHelsen, 2019-12-10:
@peterbarker btw, the panic() function works quite well, it certainly caused me some panic that I did something wrong as the simulator kept crashing. Turned out it was something simple and for the good cause. I'm glad I could help with the development.

peterbarker, 2019-12-10:
On Tue, 10 Dec 2019, RoelHelsen wrote:

> @peterbarker So the goal is to find all missing mission items in the switch statement?

Indeed - well, all the ones we support.

Obviously I thought I had them all - I was wrong :-)  Maybe I still am.

Peter

PR #13021 (2019-12-10): AP_Mission: add string for DO_MOUNT_CONTROL
Closes #13018