ArduPilot/ardupilot #13033: UC Node 53 Down! Error report from CAN device that does not use the dynamic nod

pompecukor, 2019-12-11:
### Bug report

We are getting random UC Node 53 Down! reports on Mission Planner (only while on the ground) with recent build of Arduplane.

Discussion with Andrew Trigdell suggested that the CAN packet loss threshold might need to be raised, as right now it will report on a single packed.

Side not is that the UAVCAN gps/compass producing this random error report is has been in use by us in the past 18 months, however the error report it new. That leads me to believe that it is cause by a recent change in the master code. Something done since about October. Prior to that we never got this error.

This particular hardware does not use dynamic nod.

Also platform usability and flight at not affected in any way.

_Please describe the problem_

**Version**
Latest master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

[image]

[image]

bugobliterator, 2019-12-23:
The message is printed out if Ardupilot doesn't receive a response to NodeInfo request from the UAVCAN module. Does your module support NodeInfo message? Which GPS module are you using?

WickedShell, 2019-12-23:
@bugobliterator this is known to happen with Here2 (I've produced that behavior here). We need to accept more then 1 request failing before we print scarier messages.

bugobliterator, 2019-12-23:
@WickedShell yes I agree with that, but the frequency is too high in this issue for it to be just a frequency issue.

WickedShell, 2019-12-23:
@bugobliterator It's not that rare an occurrence for me, I get something on the order of 1 every 2 to 5 minutes, something within that timescale, without time data printed above I'm not sure how you can assess the frequency (unless I've missed something)?

pompecukor, 2019-12-23:
Wicked shell is right. We only get this message 1x per operation/flight. However we get it on all our platforms with nothing actually being wrong.
All we do is scare the hell out of our customers we the random message.

bugobliterator, 2019-12-23:
I was just judging based on the amount of spams received before the GPS lock was received after GPS was detected. I could be wrong, so @pompecukor will have to post the frequency of these messages being received.

pompecukor, 2019-12-23:
It was actually after the GPS was received. What you see after that was a compete reboot. You know the message go from down to up. Everything above the "initializing APM" is after the reboot.
So this Node down message came after everything was 5/5 and ready to fly.

peterbarker, 2020-01-07:
This warning was killed.  Marking it devcall to see if anything else needs to be done....

tridge, 2020-01-11:
fixed in master

closed 2020-01-11 by tridge