ArduPilot/ardupilot #13054: Requesting Solutions for BBBlue and Linux OS with ArduCopter for a Quad+

silver2row, 2019-12-16:
### Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when I have to start from scratch during each boot on my BBBlue. It is a cyclic issue I am coming across:

1. I start the .service file for arducopter.
2. I have to bind the TX/RX.
3. I have to calibrate the ESCs.
4. If I turn off the power to my BBBlue, the arducopter.service file ceases to startup. 
5. Then, I must run the first three commands from above again no matter what happens or the system does not work.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

1. Hello...I would like to be able to start my Quad+ with the arducopter software on it and run it when I boot the Linux OS on my BBBlue. It is like the start of the Linux OS and the .service file is interrupting my ability to fly w/out binding and calibrating the ESCs each time I boot the flight controller (BBBlue).

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

I thought of ideas but I am reaching out now because I am not that involved in software right now. If you point in the correct direction, I might be able to come up with a solution. For now, I am shooting in the dark on this idea but I might be able keep looking with the proper direction.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

Hello again...I would show the video of my board working and then doing nothing but I do not think it would help. I also think the cyclic issue of having to calibrate the ESCs and rebind causes too much time away from flying the UAV/Drone/Aerial Bot. 

Seth

silver2row, 2019-12-16:
Hello Again,

I read that the reset line on the mcu needs to be set high before the pins are driven. Is this the reason behind the cyclic issue I am encountering?

Seth

khancyr, 2019-12-16:
hello,

We don't provide service to launch ardupilot, could you share the instruction you used to setup your BBblue ?
have you check that you aren't writing the configuration file and eeprom.bin file in the /tmp directory ? or into another temporary directory ?

silver2row, 2019-12-17:
Hello,

Thank you for replying. Here are the set up instructions:

    sudo apt-get install g++ make pkg-config python python-dev python-lxml python-pip
    sudo pip install future
    git clone https://github.com/ArduPilot/ardupilot
    cd ardupilot
    git branch -a  # <-- See all available branches.
    git checkout Copter-3.6  # <-- Select one of the ArduCopter branches.
    git submodule update --init --recursive
    ./waf configure --board=blue  # <-- BeagleBone Blue.
    ./waf
    sudo cp ./build/blue/bin/a* /usr/bin/ardupilot

I got this info. from this page: https://github.com/imfatant/test.

Seth

P.S. I will check the eeprom.bin file in the Debian OS I have on my BBBlue. I have not checked the /tmp dir. recently. I will go and check it.

silver2row, 2020-01-02:
Hello,

Anyway...thank you.

Seth

closed 2020-01-02 by silver2row