ArduPilot/ardupilot #13064: Flight mode change failed

marselomeri, 2019-12-17:
Flight mode change failed , trying to switch from **Guided** to **Auto** mode. 
I am trying to have a test flight in a SITL environment, after the quadrotor  takes off, it can not switch to auto mode to complete the mission (waypoints flight plan). 

I am using Copter 4.0
I upload the flight plan and give the commands using MAVproxy ground station. 

Any help would be appreciated!

peterbarker, 2019-12-17:
On Tue, 17 Dec 2019, marselomeri wrote:

> Flight mode change failed , trying to switch from Guided to Auto mode.

Logs?

Error messages?

marselomeri, 2019-12-17:
> On Tue, 17 Dec 2019, marselomeri wrote: Flight mode change failed , trying to switch from Guided to Auto mode.
> Logs? Error messages?

Here are all log files, and Error message is "APM:Flight mode change failed" at the console module. 

[log files.zip](https://github.com/ArduPilot/ardupilot/files/3975891/log.files.zip)

rmackay9, 2019-12-18:
This is really a support question so could you raise a report in the Copter-4.0 discussion area? https://discuss.ardupilot.org/c/arducopter/copter-40

If possible also include a dataflash log which has much more detail about the vehicle's status.

Sorry to be strict but otherwise our to-do list turns into a support forum

closed 2019-12-18 by rmackay9

peterbarker, 2019-12-18:
On Tue, 17 Dec 2019, marselomeri wrote:

> Here are all log files, and Error message is "APM:Flight mode change failed" at the console module.

This isn't ArduPilot :-)

Seems you're running Icarous.  You might contact NASA about that :-)

peterbarker, 2019-12-18:
.... also, those logs don't contain the error string you originally pasted in.

auturgy, 2019-12-18:
I’m not NASA, but to clarify, the default implementation of Icarous uses Ardupilot as the flight controller. The extra params are force loaded via MAVProxy, for Icarous configuration. Not a common approach, but I don’t hate it.

Regards,

James