ArduPilot/ardupilot #13074: Add RELAYx_OPTION?

rmackay9, 2019-12-18:
@lthall mentioned that he would like to use a relay input to trigger Copter's stand-by state (used to allow a secondary flight controller to be inactive until it's needed for failover).

At the moment I think we have multiple features that can read from a relay including the wheel encoders, RPM sensor and camera trigger.  We wonder if perhaps we should create a relay input manager a bit like the serial manager or RC Input library's auxiliary switch code.  So we would have RELAYIN_OPTION which would specify how the relay is being used instead of various features having _PIN parameters.

In fact, it doesn't really matter which way we do it (feature->pin or pin->feature) but at some point it requires less parameters if we use the pin->feature method.

WickedShell, 2019-12-18:
Pedantic: It's not a relay if it's an input, it's just reading a GPIO :)

We have a really bad pattern with the existing `RELAY_` parameters where the parameters map a virtual relay number to a pin, then a library uses the relay number to index it. For example both camera library and servo relay events use a hard coded relay 0, which means you can't use both of them, and don't accidentally try and use that for parachutes by accident :)

The problem is the other direction a ton of pins can be used (basically any pin that is broken out can be used, particularly if you don't have a sensor connected to it, which would be a massive number of pins to define. The only good news with that approach is you can't accidentally use the same pin in two different paths, but given the param explosion problem (and the problem of labeling the pins meaningfully for the user) I don't think this is the right path, and adding PIN parameters where need would be better.

IamPete1, 2021-08-15:
closed by https://github.com/ArduPilot/ardupilot/pull/15484 it can also do the standby use case mentioned here.

closed 2021-08-15 by IamPete1