ArduPilot/ardupilot #13075: MOT_SLEWRATE limits

jimovonz, 2019-12-18:
### Bug report

The actual slew rate achievable using the MOT_SLEWRATE parameter does not seem to correspond with stated range of 0-1000 percent throttle per second. The line in the code found [here](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/AP_MotorsUGV.cpp#L218):
    const float throttle_change_max = MIN(1.0f, (float)_slew_rate * dt);
 limits the minimum delta to 1 percent of throttle per loop cycle. For the Rover default of 50Hz, this equates to 50% throttle per second - any parameter values less than 50 results in the same slew rate of 50. I do not see the intended purpose of this line of code as there is already check for parameter values of zero or less and it certainly does nothing to enforce an upper limit. Perhaps this line should read: 
    const float throttle_change_max = MIN(1000.0 * dt, (float)_slew_rate * dt);

**Version**
Rover-4.0

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

**Airframe type**
NA

**Hardware type**
NA

**Logs**
NA - reproducible in SITL using current master

jimovonz, 2020-05-20:
This bug is still present in master...

jimovonz, 2020-07-17:
Pull request: https://github.com/ArduPilot/ardupilot/pull/14843

IamPete1, 2021-07-26:
fixed by https://github.com/ArduPilot/ardupilot/pull/14890/files

closed 2021-07-26 by IamPete1