ArduPilot/ardupilot #13076: QGC mechanism for Service Bulletin notification

DonLakeFlyer, 2019-12-18:
As it currently stands both MP anbd QGC notify the user if the stable firmware they are running is older than the latest stable. I don't know how MP does that but QGC does it by downloading the following file: ```http://firmware.ardupilot.org/[Plane/Copter/...]/stable/PX4/git-version.txt```. Parsing it and comparing version numbers.

There also needs to be a mechanism to notify the user if the current stable firmware they are running has a Service Bulletin against it and they should upgrade immediately.

Given the current directory structure of the firmware server stable builds which looks like this:
```http://firmware.ardupilot.org/[Plane/Copter/...]/stable-#.#.#```

I propose that a GCS should attempt a download in the appropriate stable dir for a file named ```SB.txt```. So for example to report a Service Bulletin against Plane 3.9.8 a you add the following file: ```https://firmware.ardupilot.org/Plane/stable-3.9.8/SB.txt```

The contents of the SB.txt file would contain the Service Bulletin details. The non-existence of the file would indicate no Service Bulletin against that version.

DonLakeFlyer, 2019-12-18:
The other possibility is to have things like SB-enu.txt and so forth for translated versions.

khancyr, 2019-12-18:
I think that is an excellent idea

Pedals2Paddles, 2019-12-18:
Excellent idea!

I'm thinking the main challenge here will be access control to the FW server.   For example, Hex I assume would want access to the cube build directories, but should not be allowed access to any of the others.  And nobody else should have access to the Hex cube build directories.

Are any hardware manufacturers besides Hex even issuing service bulletins?  I can't say I've ever seen a Holybro Service Bulletin or mRo service bulletin, or Banggood service bulletin out there.  To my knowledge, Hex is the only manufacturer being proactive and doing service bulletins.  So that would make access control initially very simple.  But, in the unlikely event other manufacturers start being proactive, access control will get more complicated.

The alternative is for the manufacturers to submit their service bulletins to someone on the dev team to evaluate and upload to the appropriate directories.  This solves the access issue, but of course requires someone to actually take on that role.

DonLakeFlyer, 2019-12-18:
This proposal isn't for hardware service bulletins. That would be a difference thing. It's for firmware service bulletins.

auturgy, 2019-12-18:
My 2c - Agree with Don, but there’s also an area of overlap: I’d suggest that if the fix to an issue is a firmware update, the gcs is a good tool for the alert, regardless of the underlying cause. A useful test case might be a SB on Pixhawk1-1M/fmuv2 to update bootloader. It’s harmless but also has benefits.
As for who releases the SB: I’d suggest setting up a folder in GitHub and using the normal git workflow, to ensure they get reviewed and are centrally controlled. This should be either standalone or in the firmware repo’s, not gcs repo’s, to ease pickup by the various gcs’s.

Pedals2Paddles, 2019-12-18:
Maybe we are talking about the same thing. ArduPilot does not issue service bulletins now (although maybe we should but that's a whole other discussion). The only service bulletins I've ever seen are from Hex, who is a hardware manufacturer. Their service bulletins happen to be related to firmware though so that's what I'm thinking of here.

I agree that if a manufacturer has a service bulletin that is a specifically hardware related thing, like hey we found a defect in our manufacturing, this may not be the avenue for that.

peterbarker, 2019-12-19:
On Wed, 18 Dec 2019, Don Gagne wrote:

> As it currently stands both MP anbd QGC notify the user if the stable
> firmware they are running is older than the latest stable. I don't know
> how MP does that but QGC does it by downloading the [...]

They're using our manifest file - 
http://firmware.ardupilot.org/manifest.json.gz

I am strongly opposed to browsers parsing directory structures.

> There also needs to be a mechanism to notify the user if the current 
> stable firmware they are running has a Service Bulletin against it and 
> they should upgrade immediately.

Adding more data to the manifest.xml has been done several times already 
without breaking existing users - so I think we ought to find some way to 
add the service bulletin information for those vendors that do that sort 
of thing.

Adding some sort of flag indicating a version should not be used would 
seem like a a natural thing to put in the JSON.

DonLakeFlyer, 2019-12-19:
Using the manifest file requires the gcs to know the board type. This is not possible in all cases.

Pedals2Paddles, 2019-12-19:
Still wondering which service bulletins from whom this would be regarding?

peterbarker, 2019-12-20:
On Thu, 19 Dec 2019, Don Gagne wrote:

> Using the manifest file requires the gcs to know the board type. This is not possible in all cases.

"Don't fly this" may only apply to a specific board, 'though.

If we do need to key off something else we could add another top-level 
section in parallel to "firmware" - what would it be called, what would 
the keys be?

tridge, 2019-12-23:
note that stable/PX4/git-version.txt will disappear completely as soon as Copter-4.0 is released. It is already gone for plane and rover

tridge, 2019-12-23:
What MissionPlanner has done is to add a link to display the release notes. Using these file:
https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/release-notes.txt
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/ReleaseNotes.txt
https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/release-notes.txt
https://github.com/ArduPilot/ardupilot/blob/master/ArduSub/ReleaseNotes.txt

tridge, 2019-12-23:
for seeing if the user has the latest firmware the manifest.json is the preferred method

CraigElder, 2019-12-23:
Also Antenna Tracker
https://github.com/ArduPilot/ardupilot/blob/master/AntennaTracker/ReleaseNotes.txt

Pedals2Paddles, 2019-12-23:
Release note is a much different thing than service bulletins though.  Linking to release notes is pretty simple since they already exist as a file in our github for each vehicle. Service bulletins are notices sent out by [WHO] for any number of reasons critical to the operation of the firmware or hardware combinations.

DonLakeFlyer, 2019-12-25:
> for seeing if the user has the latest firmware the manifest.json is the preferred method

I've noted that and will change QGC with Stable 4.1

peterbarker, 2020-11-25:
We've implemented the "User Alert" system for this: https://ardupilot.org/plane/docs/common-user-alerts.html

MAVProxy's implementation is here: https://github.com/ardupilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_useralerts.py#L1

closed 2020-11-25 by peterbarker