ArduPilot/ardupilot #13085: MavFTP: stucked

liang-tang, 2019-12-20:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

step by step:
1.clean up waypoints.
2.reboot.
3.use CTRL+X to Show FTP window.
4.delete whole TERRAIN directory.
5.reboot.
6.write waypoints. (remake TERRAIN directory)
7.use CTRL+X to Show FTP window.
8. nothing we can get from APM directory and also FCU reboot probablly.

But if  exchange step 6 and 7,  no issue occur.

**Version**
master branch today
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ * ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
defalut after reset
**Hardware type**
mindpx-v2
**Logs**
no log

WickedShell, 2019-12-21:
I can't reproduce this with a cube.

liang-tang, 2019-12-23:
> I can't reproduce this with a cube.

I'll try it again.

liang-tang, 2019-12-23:
there is a video that reproduce with cube.
MP: 1.3.69
ArduPlane: V4.1.0dev (19c02471)
https://drive.google.com/open?id=10vKB2JaKgDaKCpUyhL3DW2G_Hw0QPcJB

WickedShell, 2020-02-10:
@rmackay9's steps to repro (from #13536):
  1. flash master onto the Cube Black (I was using "ArduCopter V4.0.3-dev 0cf747c")
  1. restore all parameters to their defaults
  1. open Mission Planner and press Ctrl-X to connect using MAVFTP
  1. attempt to open the APM directory

rmackay9, 2020-02-10:
I've reproduced this in master, Copter-4.0.1 and Copter-4.0.2-rc4

WickedShell, 2020-02-11:
Repros on plane as well.

rmackay9, 2020-02-11:
This is fixed in master and is included in Copter-4.0.2

@liang-tang, thanks very much for the report!

liang-tang, 2020-02-23:
SITL with plane 4.0.5beta1
Stack overflow in thread FTP

[image]

peterbarker, 2020-02-23:
Stack trace would be good.  You will need to enable the operation according to the instructions there.

Steps to reproduce would be great!

rmackay9, 2020-02-24:
@liang-tang,

Thanks very much for the follow-up.

By the way, I tried to reproduce this on Copter (latest) but wasn't able to.  If it can be reproduced with Copter-4.0.3-rc1 then I'll stick it back on the [Copter-4.0 issues list](https://github.com/ArduPilot/ardupilot/issues/12547).

I think with Copter-4.0 we've done a good job of tracking and resolving bugs so getting it on this list is another good way to ensure it's not forgotten.

liang-tang, 2020-02-24:
The code is at /home/ubuntu/My_ArduPilot, and switched to Copter-4.0.3-rc1 branch.
Setp 1, 
in direction /home/ubuntu/My_ArduPilot, execute
>  ./Tools/autotest/sim_vehicle.py -v ArduCopter

Setp 2,
use CTRL+X to get content, then stack overflow.

**BTW**,
after increased stack size of ftp thread to 8k(just have a try), ftp works well.
and I got the root direction of ubuntu(version 16.04)
[image]
It may be that too many files are read, causing a stack overflow. Stack usage is positively correlated with the number of files read.

peterbarker, 2020-02-24:
Thanks @liang-tang !  Reproduced on ArduCopter-beta, but does not trivially reproduce on master.

@rmackay9 we might be missing a patch into the Copter beta?

liang-tang, 2020-02-24:
@peterbarker yes, master works well, tried just now.

liang-tang, 2020-02-24:
@peterbarker  we read ardupilot direction on master, but read root direction on Copter beta, different direction.

peterbarker, 2020-02-24:
f504009287221c7fda9cfa9c389a13af702aa9fd is probably the missing patch

We don't run the check-stack stuff on master because the pthread-attr-set-stack call we make doesn't work correctly.

Plane beta is also missing the patch.

rmackay9, 2020-04-22:
I think we can close this issue because it's fixed in master.  Just as a side note, getting the fix into a stable branch is a separate thing not related to having the issue open.

Davidsastresas, 2020-07-02:
I just found this thread because I got this issue reported by SITL in ubuntu 20.04. In 4.0.5 apparently the f504009 patch was not present and the error was there, I cherry-picked the patch and now it is working nicely.

rmackay9, 2020-07-02:
@Davidsastresas, txs for the feedback.  I think this patch is included in Plane-4.0.6 that is available for beta testing.  Plane isn't my baby but it should be there.

peterbarker, 2020-07-02:
That patch is not in the current ArduPlane beta.

Davidsastresas, 2020-07-04:
It would be cool to include it, the patch has been there since January!

rmackay9, 2020-07-07:
@Davidsastresas, Tridge has confirmed that this is included in the Plane beta so if you get a chance to test with the latest beta that would be good, txs again!

peterbarker, 2020-07-07:
@Davidsastresas so the patch isn't there but the code is.  Are you able to reproduce this bug on either of Copter or Plane betas?

Davidsastresas, 2020-07-07:
I just checked both, copter and plane betas and it works so the patch should be there. Thank you!